bacnet-stack/bacnet-stack #5: Add cmake support

gocarlos, 2019-12-18:
Cmake is the most used build system which is supported by most ides now (vscode, qtcreator, visual studio, clion, etc)

It is now also natively supported on windows with visual studio (open project and it works)

It would be great therefore to add cmaks support to this project
We could sponsor this feature

@skarg what do you think?

gocarlos, 2019-12-18:
We also use the conan package manager, cmake would help a lot to make the packaging easier

skarg, 2019-12-19:
I'm okay with adding cmake as long as there is an easy way to maintain it and verify it is working (i.e. via continuous integration).

closed 2019-12-19 by skarg

yegorich, 2019-12-23:
@gocarlos what is the reason for CMake version restriction. As far as I can see, you don't any special features provided by 3.12. I have no problems building the stack using 3.10 on 18.04 LTS (Bionic Beaver).

gocarlos, 2019-12-23:
> . I have no problems building the stack using 3.10 on 18.04 LTS (Bionic Beaver).

from my side, we can lower it to 3.10 which is shipped with 18.04... 

PS: I often do not use the cmake from the debian packages but the one from pip where you get always the latest versions

gocarlos, 2019-12-23:
@yegorich https://github.com/bacnet-stack/bacnet-stack/pull/11 lowers cmake min requirement to 3.10