bacnet-stack/bacnet-stack #40: Higher Level Abstraction

digitaldex, 2020-01-24:
Hello, are there any plans for higher levels of abstraction for your stack? Something like a communication object for client/server use with methods for reading/writing data or subscripe covs. Its really hard to start working with your stack. All demos work fine but there is no starting point for using the stack in own applications.

gocarlos, 2020-01-24:
I agree with you, eventually we could make a object oriented C++ wrapper class for a client and one for a server.

digitaldex, 2020-01-25:
I would like to see something like that but i dont think that i could be helpful building it. This monster here is too complex for my programming knowledge. But thats the point in this issue. Steve did an incredible job coding this but i assume there are not a lot of people who could make use of it. Something is missing that glues all of this here together to make it more useful for a wider group of people.

tblong, 2020-06-28:
@digitaldex, Agreed.  I would also like to know if anyone has undertaken exporting all or parts of this library so it could be used from a managed framework like .NET.

skarg, 2020-07-01:
There are already open BACnet client libraries:
[.NET C# - YABE](https://sourceforge.net/projects/yetanotherbacnetexplorer/)
[Python - BACpypes](https://github.com/JoelBender/bacpypes)
[Java - BACnet4J](https://github.com/infiniteautomation/BACnet4J)
[NodeRed](https://flows.nodered.org/node/node-red-contrib-bacnet)
[JavaScript](https://github.com/fh1ch/node-bacstack)

Specifically about BACnet C# and .NET: this BACnet C library requires a lot of marshaling and other arms length methods to work within C# applications.  Development becomes much easier for C# and .NET developers when using a native library such as [YABE](https://sourceforge.net/projects/yetanotherbacnetexplorer/) project.

For this BACnet C library, are some uses alongside C++ or objective-C, but the library is mostly focused on embedded systems and command line tools.