bacnet-stack/bacnet-stack #44: Python wrapper for BACnet/MSTP

ChristianTremblay, 2020-02-13:
Hi ! I'm wondering if I could have some help from you to build a C Extension from the stack to allow Python to use MSTP.

Speed and timers being at the core of MSTP, it's very hard (let say it won't be possible) to have MSTP working in pure Python... but it would be possible as a C extension.

My C knowledge is -2 on a scale of 0 to 10... but I could help with the Python part.

[ref] https://docs.python.org/3/extending/building.html

That would be a super nice add-on to our projects.

https://github.com/ChristianTremblay/BAC0
https://github.com/JoelBender/bacpypes 

Thanks !

ChristianTremblay, 2020-04-14:
Seems someone else made it
https://github.com/riptideio/misty

closed 2020-04-14 by ChristianTremblay

deve87, 2024-04-03:
Is it possible to use misty directly in BAC0?