bacnet-stack/bacnet-stack #48: do new release to github

gocarlos, 2020-02-18:
some build systems normally take tags and not commits e.g. yocto build system, conan package manager

It would be great if we could make a e.g. patch release as now we have cmake support and can use this to be cross platform.

gocarlos, 2020-03-11:
@skarg what do you think about that?

skarg, 2020-03-11:
I had detailed the tasks that I thought we should complete for a 1.0.0 release via a [message on our mailing list](https://sourceforge.net/p/bacnet/mailman/message/36761701/). Most of these are completed. I've really only worked with [GitFlow work flow](https://nvie.com/posts/a-successful-git-branching-model/) but there are other git branching models such as [Github flow](https://guides.github.com/introduction/flow/). Is there a particular workflow that you are expecting here?

gocarlos, 2020-03-18:
My request is more of a general release tag, can be 0.9.0 or so, no need to be 1.0.0

I like the github workflow, master should be deployable to production. 
Other than that i think that in 2020 we should go into a `semi rolling release` mode as chrome and firefox is doing. E.g. you have versions but normally stuff is fixed in the next version e.g. 1.0.1 (backwards compatible, patch release -> no api break)

This lower the maintenance overhead and makes sure that fixes are delivered in a timely manner... this is more and more important for connected IoT devices

skarg, 2020-03-18:
There were already 0.8.6 and 0.9.1 tags or branches in SVN, so the next release will be 1.0.0 version. I don't want to go to 0.10.0 or 0.11.0 etc.  

Using a semi-rolling release usually causes issues with API and is generally accepted that API may occasionally break. Since these releases are usually deployed in embedded products with long life cycles, I think stability is expected. I suppose that anyone wanting long term stability in their version can continue to patch and tag.

gocarlos, 2020-03-18:
ok, that good...
question: shouldn't we focus on github and port missing tags to here? otherwise we have stuff on svn and stuff on git...

regarding version 1.0.0: is bacnet secure connect still a thing for version 1.0.0? are the others completed?

- [x] port to git
- [x] fix vulnerabilities  
- [x] change folder structure to lower name collisions
- [x] indentation and code style -> clang-format
- [ ] bacnet secure connect datalink

skarg, 2020-03-19:
I will not be able to have BACnet/SC ready for 1.0.0 version. I'm actively working on a project that will use it, and one of my tasks is to develop BACnet/SC, but I haven't done anything but planning.

Most SVN tags and some branches were already exported to git and exist. I see there were no tags for 0.9.x, so as you say, we should port them here (tag the commits that are already in this git repo).  The SVN release branches were not exported as branches but as folders under git releases branch. We should tag those and purge the branches.

skarg, 2020-04-29:
Version 1.0 release checklist:
- [ ]  port to git
- [ ]  fix known vulnerabilities
- [ ]  change folder structure to lower name collisions
- [ ]  indentation and code style -> clang-format in src and apps and ports
- [ ]  tag 0.8.7 and create and add restrictions to bacnet-0.8 branch
- [ ]  tag 0.9.1 and create and add restrictions to bacnet-0.9 branch
- [ ]  tag 1.0.0 and create and add restrictions to bacnet-1.0 branch
- [ ]  generate Windows exe files from apps and place into zip for release on bacnet.sf.net

closed 2020-06-22 by skarg