bacnet-stack/bacnet-stack #49: Building shared bacnet-stack libraries using cmake fails + finopenssl

madebr, 2020-02-18:
Hello,

While helping to package this project at https://github.com/conan-io/conan-center-index/pull/910, I noticed some problems with bacnet-stack.

- When building shared libraries using cmake, the build fails.
I think this is because the linux build has not all files included.
- Also, some files use `OpenSSL`, but the cmake script does not look for it.
It requires a `find_package(OpenSSL)` somewhere + use of the variables/targets.

Fixing these issues should also fix some potential problems with the static build.

travis output
https://travis-ci.com/madebr/bacnet-stack/jobs/288603211
changes to `.travis.yml`:
https://github.com/madebr/bacnet-stack/commit/42041220cf1dc864614beadfa1be645b28cde550

gocarlos, 2020-02-18:
yes, I also realized now that one header of the bacnet-stack has dependencies to openssl...
Though I do not see any other reference to openssl in the old build system e.g. makefile or msbuild... 

@skarg is this code with openssl still supported? e.g. `bacsec_linux.c`

I tried to bring now the openssl dependency to the code to make it compile though the openssl functions are already deprecated since 1.1.0: `HMAC_CTX_init() was replaced with HMAC_CTX_reset() in OpenSSL versions 1.1.0.` https://www.openssl.org/docs/man1.1.0/man3/HMAC.html 
there are some new functions as a drop in placement, other functions were removed and the code has to be adapted or if not needed anymore removed...

skarg, 2020-02-18:
The recent BACnet/SC addendum also included deprecation of Clause 24 security, so bacsec_linux.c and the openssl dependency will likely be removed.  Technically it can still be used for older protocol-revisions.

gocarlos, 2020-02-28:
https://github.com/bacnet-stack/bacnet-stack/pull/52 should fix this problem...

only the msvc linker seems to have issues with paths/spaces....

madebr, 2020-02-28:
@gocarlos I think the error is about a missing implib.
This is because of the same reason as https://github.com/conan-io/conan-center-index/pull/968#issuecomment-591455195.
When no `__declspec(dllexport)` annotations or no `def file` is given, the linker assumes all functions are internal and all are put in the dll.
What `__declspec(dllexport)` does is telling the linker that this function should be visible to users through the implib, which is a `.lib` file (for MSVC, for mingw it is a `.a`).

Check this cmake module out:
https://cmake.org/cmake/help/latest/module/GenerateExportHeader.html

gocarlos, 2020-02-28:
yeap, lets fix the openssl stuff first, then the shared lib thing...

@skarg do you want to support the bacnet-stack as a shared library under windows?
This would need the exports as @madebr mentioned above

skarg, 2020-03-02:
I have been building the command line tools under Windows using MinGW, either as archive (static lib) in previous (0.8.x/0.9.x) or as linked objects (0.99.x-->1.0.0).  If a particular Windows build needs the `__declspec(dllexport)` added somewhere for other compilers or build systems, I'm okay with someone adding it as long as it is not breaking the other builds.

PR #54 (2020-03-06): Export symbols in order to support shared libraries
Fixes #49

This pr does the following:
- fix warnings bacause the definition `BAC_ROUTING` was not passed
- reorder cmake a bit in order to use less variables and do all port specific work in one place.
- Support both openssl 1.0 and 1.1 
  * I didn't see encryption support for mac and win, so I used the openssl code for mac and win also.
- When building a shared library, hide all symbols by default (using `-fvisibility=hidden`). This is the same behavior as MSVC. A symbol can be exported by prepending it with `BACNET_STACK_EXPORT`.
  I did n [... cut]
closed 2020-03-06 by skarg