bacnet-stack/bacnet-stack #59: add BACnet Secure Connect Datalink

gocarlos, 2020-03-18:
this issue shall track the status of the `secure connect datalink` implementation

hschrankQmisoft, 2021-03-12:
is there any development in progress?

stevedh, 2022-03-30:
I have a working branch here, if anyone wants a look or wants to contribute. 

https://github.com/stevedh/bacnet-stack/commits/feature/sc

skarg, 2024-09-09:
Due to the size of the pull request branch changes in [#423] at github, I have been refactoring the non-BACnet/SC related changes and placing them into their own pull-requests to shrink the size and only include relevant changes and additions. The BACnet/SC functionality in the pull-request branch, however, is complete and has been validated with various vendors.