bacnet-stack/bacnet-stack #68: Port ESP32: Missing file bacport.h

waterfox207, 2020-04-13:
Hello,

I would like to compile bacnet for my ESP32. But I'm getting an error, that bacport.h is missing. 
Where do I get this file?

skarg, 2020-04-13:
bacport.h was designed to accommodate subtle differences between OSs Windows, BSD, Linux and for the example apps. Which file includes bacport.h that ESP32 is using? Perhaps it can be removed?

waterfox207, 2020-04-13:
Okay. I put the win32 bacport.h into the project and deleted all the windows header files. 
Now I have some warnings. Wich I think are for now not important. 

But the compilation fails again with two missing files: uci_config.h and uci.h.

skarg, 2020-04-13:
UCIX includes uxix.h, but only if you define BAC_UCI - are you BAC_UCI  defining somewhere in your build or project?

waterfox207, 2020-04-15:
I'm trying to compile the port for the esp32 without any changes. 
Where do I define or undefine BAC_UCI?  
By the way what does uci?

skarg, 2020-04-15:
The abbreviation UCI stands for [Unified Configuration Interface](https://oldwiki.archive.openwrt.org/doc/uci) and is intended to centralize the configuration of OpenWrt.

The BACnet source files use the construct `#ifdef BAC_UCI` which is only including the files if BAC_UCI is defined somewhere in either the build or project files, or within some header file.  I don't expect that you would need UCI for your project - so you need to figure out where in your project it is defined and remove the define.

waterfox207, 2020-04-15:
i found the bac_uci only as ifdef in the divice.c file 
[image]

It's also not defined in the sdkconfig file. I really don't know where it could be defined?

waterfox207, 2020-04-16:
An small updated from today.
I think that the problem is related to the auto generation of CMakeLists.txt from platformio. Because it sets all included files to the the CMakeLists.txt. Since I put the whole Bacnet folder into the esp32/src folder. That could be the reason why platformio tries to compile BAC_UCI?

So I've tried to compile it directly with ESP IDF V4.1. 
But this runs me into the following warnings and errors: 
[image]
[image]

Maybe someone can help me to bring this port bac to life.

completehvac, 2020-04-19:
I am also trying to get this to work currently, I downloaded the code from a different source, bacnet-esp-idf-master, I have been able to compile the code, and can read values with no problem, however I cannot write any values, the LED value or any others I create are always null on all priorities except fallback which is 0.
Any ideas why this would be occurring?

pdu_len = datalink_receive(&src, &Rx_Buf[0], MAX_MPDU, timeout);
            if (pdu_len) 
            {                
                npdu_handler(&src, &Rx_Buf[0], pdu_len);

                if(Binary_Output_Present_Value(0)==BINARY_ACTIVE)
                    gpio_set_level(BACNET_LED,1);
                else
                    gpio_set_level(BACNET_LED,0);  
				
				if(Binary_Output_Present_Value(1)==BINARY_ACTIVE)
					printf("\n Bin True");
				else
					printf("\n Bin False");
            }

the monitor always prints "Bin False"

waterfox207, 2020-04-21:
thank you for your response.
with your source it seems like it's runs on my esp32, too.
Since I don't have any BACnet IP device I'm not able to test it at the moment. 

I only have BACnet mstp devices so I will try to port the MSTP protocol to the code. Maybe someone is here who can help me with this project or can give me some tips how to realize it.

completehvac, 2020-04-22:
waterfox, I look forward ot hearing how you get on, the code runs as you say, but bacnet is not operating correctly.

completehvac, 2020-05-02:
@waterfox207 Did you make any progress?
@skarg Can you shed any light? It seems the example allows to read data values, but I cannot write any values, they never get updated and are always null, except lowest priority which is always 0.

waterfox207, 2020-05-02:
Not really. 
With the bacwi on my pc an were able recognize the esp in my network. 
If also edited the RS485.c for implementing mstp. But I think that there are some problems with the source we are using (the bacnet-esp-idf-master). 

So I have tried again to compile the current master with esp idf. Sadly with no success. At the moment I'm struggling with the bip_init.c. In older versions there was also a bip.c in the stack wich now misses. 

So functions like bip_receive() are not defined. 
@skarg  are you able to help me with redoing the bip_init.c for the esp32? So we can bring this port back to live?

completehvac, 2020-05-02:
I have done a little more debugging! In my case it seems there is no receive handler functionality within the code, in the file device.c, therefore it seems no writes of data are working. I am going to look further to see where we are reading from as I assume some variables will need to be updated.