bacnet-stack/bacnet-stack #87: execute tests with github ci

gocarlos, 2020-05-17:
currently we are only building for the three major OSs, linux, mac, win32.

there are no tests beiing executed.
If tests were executed with CI, we could also get test coverage reports etc.

skarg, 2020-05-19:
I'm currently working on improving the unit testing. We used Zephyr OS and ztest. We were planning to move the tests into their own modules (from being integrated), and port ztest to this project without dependency to Zephyr OS and use in place of ctest so that we also get mocks.  The coverage using LCOV at the moment looks like this:
[image]

Did you have an alternate testing tool or path forward for test coverage report, @gocarlos ?

gocarlos, 2020-05-19:
we use gtest which also works with C libraries, this testing library (C++) is widely available (debian, conan, vcpkg) and has different reporting techniques e.g. ala java JUnit's, built-in mocking library etc. 

Then we also use cmake's built in `ctest`, this allows us to have also valgrind testing, and costum tests with arguments e.g. if you have some test data and one test binary, you can create 10 tests by using 10 different arguments for the tests e.g. `test --data 1`, example here: https://github.com/cose-wg/COSE-C/blob/master/test/CMakeLists.txt#L35

for code coverage we use `gcovr` as it has a better user interface for us than `lcov` 
though here we could also use a third party service like coveralls which is eben easier to use: https://github.com/marketplace/coveralls

closed 2022-03-09 by skarg