bacnet-stack/bacnet-stack #96: Really a bad experience of this library.

sdghchj, 2020-06-15:
No examples how to use its APIs, how to run bacserv and bacwi in README.md even all the README.* in doc directory.

gocarlos, 2020-06-17:
there are example in the folder `apps` for example: https://github.com/bacnet-stack/bacnet-stack/blob/master/apps/server/main.c

gocarlos, 2020-06-17:
though the readme and documentation can be improved, you are welcome to contribute :+1:

kal9001, 2021-07-04:
> there are example in the folder `apps` for example: https://github.com/bacnet-stack/bacnet-stack/blob/master/apps/server/main.c

Unfortunately much of this isn't obvious to 'fresh' eyes. I'm sure that to the people who have helped develop this it's all perfectly obvious and easy to work with.
But for someone trying to unfuddle the code with very little documentation... From a naive point of view the whole project looks like a rats nest of makefiles, unexplained abbreviations, mysterious functions and annoying code layout.

I can't thank the contributors enough for developing this valuable tool, but for the love of god I wish you had made better documentation. Simply saying "you are welcome to contribute" is frankly a bit rich when people can't even get started.

I know this is free software, I know you all aren't beholden to provide anything, so I apologise for the tone of my rambling, I'm just frustrated I've been looking at this on and off for three weeks now and still can't make it do anything useful!

p.s. sorry for the bump of a year old issue.

skarg, 2021-07-05:
One of the biggest challenges with managing library like this and its documentation, targeted at anything from tiny microcontrollers to PCs to VMs and used in devices, as tools, or as interfaces to larger processes, is to figure out what documentation is useful.  Nearly all the communication I have with developers or users starts with a conversation about how they would like to use BACnet in their product, on their site, or whatever it is that got their interest.  The conversation then becomes guided toward the files and tools that they can then use to do whatever it is they were wanting to do, and determining how much about BACnet does the individual understand in order to adjust the guidance.

For example, the BACnet committee has been working on Developer Guidelines, and we've focused on documents describing a BACnet library (stack), how to decode the BACnet standard and turn that into C, and how to develop a BACnet device by modeling it from the objects described in SSPC 135 Clause 12.

It would be more helpful to start the conversation by asking the questions "How would I get started doing X with BACnet?" or  "I want to do X, Y, Z with BACnet" or "What are the apps used for?" or "Why is the code so annoying laid out?"

The apps don't make much sense to someone who doesn't know anything about BACnet services, and aren't that interesting for someone who wants to make a BACnet MS/TP device using a microcontroller.

It might also be helpful if someone wants to help modernize the existing docs, or perhaps have someone suggest new document contents (maybe a title and an outline?) that tell more information, perhaps from a fresh eyes point of view.