bacnet-stack/bacnet-stack #100: make BACDL_DEFINE=-DBACDL_ALL=1 fail

sdghchj, 2020-06-23:
`make BACDL_DEFINE=-DBACDL_ALL=1` fail
`make BACDL_ALL=1` fail
`make BACDL_DEFINE=-DBACDL_ALL=1 BACDL_ALL=1` fail

So how to make it support all the datalinks?

sdghchj, 2020-06-23:
And v1.0.0 will never generate libbacnet.a ??

sdghchj, 2020-06-23:
And v1.0.0  fail when running `make`  on linux.

skarg, 2020-06-23:
Thank you for reporting the build problem with BACDL_ALL. I'll triage it and see if I can fix it for 1.0.1 for you.

The default Makefile for the apps builds on Linux. See the Continuous Integration results and their recipes and build environments in the various yml files:
[Travis](https://travis-ci.com/github/bacnet-stack/bacnet-stack/builds)
[Github](https://github.com/bacnet-stack/bacnet-stack/actions)
[AppVeyor](https://ci.appveyor.com/project/skarg/bacnet-stack/branch/master)

Is there a particular Linux distro that the build is failing on?

I'll see if I can quickly add a libbacnet.a Makefile target.  I moved away from the library style application builds for the main Makefile, but the resulting binaries are much larger now (4x or more). The previous 0.8.7 is also available in the source folder, tagged, and maintenance branch, if you want to use it until 1.x.x has that ability.

sdghchj, 2020-06-23:
My Linux is centOS7 x86-64, I remember, nothing special.

sdghchj, 2020-06-24:
make on centos
```
[root@centos bacnet-stack]$ make all
make -s -C apps all
main.o: In function `main':
main.c:(.text.startup.main+0x406): undefined reference to `datalink_cleanup'
main.c:(.text.startup.main+0x562): undefined reference to `datalink_receive'
/home/app/bacnet/bacnet-stack/src/bacnet/basic/tsm/tsm.o: In function `tsm_timer_milliseconds':
tsm.c:(.text.tsm_timer_milliseconds+0x5a): undefined reference to `datalink_send_pdu'
/home/app/bacnet/bacnet-stack/src/bacnet/basic/service/h_noserv.o: In function `handler_unrecognized_service':
h_noserv.c:(.text.handler_unrecognized_service+0x14): undefined reference to `datalink_get_my_address'
h_noserv.c:(.text.handler_unrecognized_service+0x6b): undefined reference to `datalink_send_pdu'
/home/app/bacnet/bacnet-stack/src/bacnet/basic/service/h_rp.o: In function `handler_read_property':
h_rp.c:(.text.handler_read_property+0x25): undefined reference to `datalink_get_my_address'
h_rp.c:(.text.handler_read_property+0x28c): undefined reference to `datalink_send_pdu'
/home/app/bacnet/bacnet-stack/src/bacnet/basic/service/s_iam.o: In function `iam_encode_pdu':
s_iam.c:(.text.iam_encode_pdu+0x17): undefined reference to `datalink_get_my_address'
s_iam.c:(.text.iam_encode_pdu+0x1f): undefined reference to `datalink_get_broadcast_address'
/home/app/bacnet/bacnet-stack/src/bacnet/basic/service/s_iam.o: In function `Send_I_Am':
s_iam.c:(.text.Send_I_Am+0x27): undefined reference to `datalink_send_pdu'
/home/app/bacnet/bacnet-stack/src/bacnet/basic/service/s_rp.o: In function `Send_Read_Property_Request_Address':
s_rp.c:(.text.Send_Read_Property_Request_Address+0x4d): undefined reference to `datalink_get_my_address'
s_rp.c:(.text.Send_Read_Property_Request_Address+0xe6): undefined reference to `datalink_send_pdu'
/home/app/bacnet/bacnet-stack/src/bacnet/basic/service/s_whois.o: In function `Send_WhoIs_To_Network':
s_whois.c:(.text.Send_WhoIs_To_Network+0x19): undefined reference to `datalink_get_my_address'
s_whois.c:(.text.Send_WhoIs_To_Network+0x6b): undefined reference to `datalink_send_pdu'
/home/app/bacnet/bacnet-stack/src/bacnet/basic/service/s_whois.o: In function `Send_WhoIs_Global':
s_whois.c:(.text.Send_WhoIs_Global+0x19): undefined reference to `datalink_get_broadcast_address'
/home/app/bacnet/bacnet-stack/src/bacnet/datalink/dlenv.o: In function `dlenv_init':
dlenv.c:(.text.dlenv_init+0x18): undefined reference to `datalink_set'
dlenv.c:(.text.dlenv_init+0x72): undefined reference to `datalink_init'
dlenv.c:(.text.dlenv_init+0xab): undefined reference to `dlenv_network_port_init'
collect2: error: ld returned 1 exit status
make[2]: *** [bacrp] Error 1
make[1]: *** [readprop.all] Error 2
make: *** [apps] Error 2

```

sdghchj, 2020-06-24:
Sorry, on CentOS, I haven't `make clean` after `make BACDL_DEFINE=-DBACDL_ALL=1`, which led to `undefined reference` . 
It's ok.
The key issue is BACDL_ALL.

closed 2022-01-29 by skarg