bacnet-stack/bacnet-stack #103: How to set network number in bacserv

monexiao, 2020-07-08:
I use Network_Port_Network_Number_Set(1, 49999) to set the network number as 49999 in bacserv sample app,
and then I use YABE to explore my sample bacserv, I can find the network number is 49999 in "OBJECT_NETWORK_PORT:1" of my simulated device.

But when I use "./bacwi --dnet 49999" to request bacnet device with network 49999, I can not find any device, only find device by "./bacwi --dnet 0".

Does any one can help me to know what wrong and guide me to correct it?

skarg, 2020-07-08:
The BACnet network number is only used within a routing device (two or more network ports).  When a BACnet PDU is routed via a router, the message uses an address that includes routing information, or lack of routing information if destined for the local segment.  

Therefore, the network port number is meaningless in a non-router device.  Note: It's possible for a device to learn which network number it is occupying if there exist BACnet routers on the network.

Using the bacwi with the "--dnet" option will add the extra addressing info to send a message that will be routed to the designated network number if a BACnet router and network segment with that number exist.

monexiao, 2020-07-09:
Thank you for your kind reply.
Our project is a IoT device to provide sensor data, also providing data by BACnet/IP protocol, but not a BACnet router-device. I think that our device do not need to run as a BACnet router role.
BUT our colleague always SAY all BACnet/IP devices should to provide network number (and run as  BACnet router role) because some BACnet controller vendor have provided.
This is very confusing me since our device does not connect any BACnet device on other network ports, only be connected by BACnet controller by IP network. Why our device is considered to be the same as other products ?

skarg, 2020-07-14:
Use a Network_Number of 0 which indicates that the Network_Number is not known or cannot be determined.

Per BACnet standard ANSI/ASHRAE Standard 135-2016
```
12.56.11 Network_Number
This property, of type Unsigned16, represents the BACnet network number associated with this network.
The range for this property shall be 0 .. 65534. A Network_Number of 0 indicates that the Network_Number is not known or
cannot be determined. Writing 0 to the Network_Number property shall force the value of the Network_Number_Quality
property to UNKNOWN and allows the device to attempt to learn the network number, if possible. Writing a value other than
0 shall force the Network_Number_Quality property to CONFIGURED.
If the Network_Type is PTP or NON_BACNET, then this property shall be read-only and contain a value of 0.
This property shall be writable in routers, secure devices, and any other device that requires knowledge of the network
number for proper operation. Routers are permitted to refuse a value of 0. In that case, the write request shall result in an
error response with ‘Error Class’ of PROPERTY and an ‘Error Code’ of VALUE_OUT_OF_RANGE.
If this property is writable, then a successful write to this property shall set the Changes_Pending property to TRUE. A value
written to this property shall become effective when the device receives a ReinitializeDevice service request with a
'Reinitialized State of Device' of ACTIVATE_CHANGES or WARMSTART.
```

```
12.56.12 Network_Number_Quality
This read-only property, of type BACnetNetworkNumberQuality, represents the current quality of the Network_Number
property. If the Network_Type is PTP, the Network_Number_Quality shall be CONFIGURED.
This property shall have one of the following values:
UNKNOWN None of the below meanings.
LEARNED The Network_Number was learned via receipt of a Network-Number-Is
message with a flag value of 0 (learned).
LEARNED_CONFIGURED The Network_Number was learned via receipt of a Network-Number-Is
message with a flag value of 1 (configured).
CONFIGURED The Network_Number is configured for this port.
```

monexiao, 2020-07-15:
Thanks you reply, I use network number 0 in our project indeed, includes previous project. But still been asked why it can't be set to other numbers. 
To sum up briefly, as long as our product type is not defined as BACnet Router, is there no need to provide the function of to change Number Number ?

skarg, 2020-08-13:
"This property shall be writable in routers, secure devices, and any other device that requires knowledge of the network
number for proper operation."

A BACnet router "requires knowledge of the network number for proper operation".  The router uses the number for routing, and this property is used to configure the network number so it is required that it be writable.  

Since your device does not "requires knowledge of the network number for proper operation" - it is not required to be writable. The network number could be learned by your device and populated, but the number should not be writable. If the network number was made writable in your product, it would have no effect on the operation and only give misinformation to others.