bacnet-stack/bacnet-stack #119: Manually setting the RTS pin for RS-485

kmare, 2020-09-08:
First of all, thank you for this library, great work :)
I'm trying to to communicate with a BMT-DO4. This is working fine when using an USB RS-485 dongle. But as soon as I try to use the native RS-485 port instead of the USB dongle, the communication fails (sending a bind request, but can never bind until it timeouts). 
My understanding is that I would need to manually set the RTS pin high and low depending on when I'm transmitting or receiving a frame.
Where would be the best place to add such a change? As far as I know/understand, libmodbus did something similar as it can be seen here:
https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c

skarg, 2020-09-08:
For the Linux version of RS485, the RTS pin would need to be toggled in ports/linux/rs485.c in the RS485_Send_Frame() function before and after the write() call.
For the Windows version of RS485, the RTS pin would need to be toggled in ports/win32/rs485.c in the RS485_Send_Frame() function before and after the WriteFile() call.

parthEmbedded, 2021-02-04:
Do you need to add delay before or after toggle?