bacnet-stack/bacnet-stack #130: Stm32 port: pin remapping

ThePiGrepper, 2020-10-05:
the two default I/O in the stm32 example dont appear to change actual physical pins when requested to set or clear its values. Which physical IOs are mapped to the ones recognized by the software? And how could one change those?

skarg, 2020-10-06:
Not sure which I/O you are referring to.  The GPIO that the example uses are for LEDs and the RTS (request to send) on the RS485 transceiver.  See schematic of the daughterboard and the referenced discovery kit.  Every port to another STM32 microcontroller will likely use a different peripheral driver library and would need rs485.c and led.c modifications (if your design even has LEDs).

ThePiGrepper, 2020-10-06:
when I check the device from the pc, it appears as having two IOs, BO-0 and BO-1. I suspected initially they were Led3 and Led4 from the source code, however after changing the mapping to GPIOB(because Im trying to port to a stm32f103c8 which doesnt have that many pins from GPIOC), BO-0 and BO-1 dont appear to be mapped to any physical pin. I can modify BO-0 and BO-1, however it doesnt change the output value on any pin. Why could that be happening?

skarg, 2020-10-07:
The example bacnet_task() function in bacnet.c module handled the BO-0 and BO-1 - feel free to do something interesting with them in the task:
```
    /* Binary Output */
    for (i = 0; i < MAX_BINARY_OUTPUTS; i++) {
        out_of_service = Binary_Output_Out_Of_Service(i);
        if (!out_of_service) {
            binary_value = Binary_Output_Present_Value(i);
            polarity = Binary_Output_Polarity(i);
            if (polarity != POLARITY_NORMAL) {
                if (binary_value == BINARY_ACTIVE) {
                    binary_value = BINARY_INACTIVE;
                } else {
                    binary_value = BINARY_ACTIVE;
                }
            }
            if (binary_value == BINARY_ACTIVE) {
                if (i == 0) {
                    /* led_on(LED_2); */
                } else {
                    /* led_on(LED_3); */
                }
            } else {
                if (i == 0) {
                    /* led_off(LED_2); */
                } else {
                    /* led_off(LED_3); */
                }
            }
        }
    }
```