bacnet-stack/bacnet-stack #135: Large BSS Size

kjcantrell, 2020-10-28:
Hi All,
     Trying to implement a small server example based on ti-rtos gcc (~256k SRAM), I thought I read somewhere this stack had a very small footprint (on the order of 40k but I can't find where I got this information). So far I have enabled only the "I am" service, "Read Property" Service enabled. I really only need Analog Input objects for my demo, but even without using those i am still 190k over the BSS size in my chip. Is this target just too small to handle this stack? 

Best,

kjcantrell, 2020-10-29:
I ended up going file-by-file and reducing the array sizes for a number of things I don't anticipate I'll be needing. If this is a good idea for a feature request I'll gladly clean this up and submit it for a pull request.

BACnetEd, 2020-10-29:
K

You can also consider disabling services you do not intend to support, e.g.
WPM, COV, and Objects you do not intend to support e.g. LIGHTING etc. A
place to start is config.h.

However, we have a fork of Steve's stack were we have added quite
fine-grained control over services and objects.... and it should be in
public, but our public facing fork is not so up-to-date. LMK if you are
interested and we can perhaps arrange a copy/update the public facing repo
for you.

skarg, 2020-10-30:
The default configuration settings src/bacnet/config.h file are for the apps/ that build on Linux/Windows/BSD and typically use full size buffers for many things.  If you take a peek at some of the smaller ports (i.e. look into the Makefile), you can see some of the more modest settings for microcontrollers.  For example, from ports/stm32f10x Makefile:
```
BACNET_FLAGS = -DBACDL_MSTP=1
BACNET_FLAGS += -DBACAPP_ALL
BACNET_FLAGS += -DMAX_APDU=480
BACNET_FLAGS += -DBIG_ENDIAN=0
BACNET_FLAGS += -DMAX_TSM_TRANSACTIONS=0
BACNET_FLAGS += -DMAX_CHARACTER_STRING_BYTES=64
BACNET_FLAGS += -DMAX_OCTET_STRING_BYTES=64
# if called from root Makefile, PRINT was already defined
BACNET_FLAGS += -UPRINT_ENABLED
BACNET_FLAGS += -DPRINT_ENABLED=0
```
As for the typical footprint, those are stated in the [FAQ (frequently asked questions)](https://github.com/bacnet-stack/bacnet-stack/blob/master/doc/README.faq).