bacnet-stack/bacnet-stack #141: APDU Timeout! and  TSM Timeout!

nambiraja, 2020-12-17:
Hello,

I've built the latest source from github on Ubuntu 16.04 and trying to connect to Legrand RCU (Bacnet enabled Room Control Unit).

Ethernet: eth0 - 192.168.0.222 by DHCP.

root@Milan-One:/var/www/Milan/bacnet-stack-master/bin# ./bacwi -1
;Device   MAC (hex)            SNET  SADR (hex)           APDU
;-------- -------------------- ----- -------------------- ----
  7141    C0:A8:00:DE:BA:C0    0     00                   1000
;
; Total Devices: 1

1) APDU Timotout Error
root@Milan-One:/var/www/Milan/bacnet-stack-master/bin# cd bacnet-stack-master/bin/./bacepics -v 7141
-bash: cd: bacnet-stack-master/bin/./bacepics: No such file or directory
root@Milan-One:/var/www/Milan/bacnet-stack-master/bin# ./bacepics -v 7141
Error: APDU Timeout! (10s)
}
End of BACnet Protocol Implementation Conformance Statement

2) TSM Timotout Error:
root@Milan-One:/var/www/Milan/bacnet-stack-master/bin# ./bacwp 7141 1 0 85 16 -1 0 0
Error: TSM Timeout!
root@Milan-One:/var/www/Milan/bacnet-stack-master/bin# ./bacrp 7141 1 1 87
Error: TSM Timeout!

What should I do to get the communication working? My understanding is that this should work out of box, if compile was successful.

Thanks

skarg, 2020-12-17:
There is a good discussion in previous Discussion about finding the cause of the timeout.
[BACnet Protocol Stack/Discussion/Real Devices](https://sourceforge.net/p/bacnet/discussion/402141/thread/da0b9e71/)

nambiraja, 2020-12-17:
I tried with all mentioned steps but the same error was showing.

Following environment variables:
BACNET_APDU_TIMEOUT=12000,
BACNET_IFACE=eth0,
BACNET_IP_PORT=47808,
BACNET_BBMD_PORT=47808,
BACNET_BBMD_TIMETOLIVE=120000

root@Milan-One:/var/www/Milan/bacnet-stack-1.0.0/bin# ./bacwi -1
;Device   MAC (hex)            SNET  SADR (hex)           APDU
;-------- -------------------- ----- -------------------- ----
  7141    C0:A8:00:DE:BA:C0    0     00                   1000
;
; Total Devices: 1
root@Milan-One:/var/www/Milan/bacnet-stack-1.0.0/bin# ./bacepics -v 7141
Error: APDU Timeout! (163s)
}
End of BACnet Protocol Implementation Conformance Statement

nambiraja, 2020-12-17:
And Thanks for your prompt reply.

skarg, 2020-12-17:
Is there another service using the BACnet UDP port? What does netstat show?

$ netstat --numeric-ports --udp --listening

nambiraja, 2020-12-17:
No. Please find the below output of netstat.

root@Milan-One:~# netstat --numeric-ports --udp --listening
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
udp        0      0 0.0.0.0:34653           0.0.0.0:*
udp        0      0 0.0.0.0:32108           0.0.0.0:*
udp        0      0 0.0.0.0:55180           0.0.0.0:*
udp        0      0 0.0.0.0:60866           0.0.0.0:*
udp        0      0 0.0.0.0:45534           0.0.0.0:*
udp        0      0 0.0.0.0:60910           0.0.0.0:*
udp        0      0 0.0.0.0:52752           0.0.0.0:*
udp        0      0 0.0.0.0:36386           0.0.0.0:*
udp        0      0 0.0.0.0:53              0.0.0.0:*
udp        0      0 0.0.0.0:58425           0.0.0.0:*
udp        0      0 0.0.0.0:60999           0.0.0.0:*
udp        0      0 0.0.0.0:56926           0.0.0.0:*
udp        0      0 0.0.0.0:57958           0.0.0.0:*
udp        0      0 0.0.0.0:55409           0.0.0.0:*
udp        0      0 192.168.0.94:123        0.0.0.0:*
udp        0      0 127.0.0.1:123           0.0.0.0:*
udp        0      0 0.0.0.0:123             0.0.0.0:*
udp        0      0 0.0.0.0:54908           0.0.0.0:*
udp        0      0 0.0.0.0:25221           0.0.0.0:*
udp        0      0 0.0.0.0:55953           0.0.0.0:*
udp        0      0 0.0.0.0:32943           0.0.0.0:*
udp        0      0 0.0.0.0:38575           0.0.0.0:*
udp        0      0 0.0.0.0:46270           0.0.0.0:*
udp        0      0 0.0.0.0:5353            0.0.0.0:*
udp        0      0 0.0.0.0:48447           0.0.0.0:*
udp        0      0 0.0.0.0:36161           0.0.0.0:*
udp        0      0 0.0.0.0:33090           0.0.0.0:*
udp6       0      0 :::41920                :::*
udp6       0      0 :::53                   :::*
udp6       0      0 fe80::e43a:48ff:fe4:123 :::*
udp6       0      0 fdb9:f190:fc9b:0:e4:123 :::*
udp6       0      0 ::1:123                 :::*
udp6       0      0 :::123                  :::*
udp6       0      0 :::5353                 :::*
root@Milan-One:~#

nambiraja, 2020-12-17:
root@Milan-One:~# netstat --numeric-ports --udp --listening
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
udp        0      0 0.0.0.0:34653           0.0.0.0:*
udp        0      0 0.0.0.0:32108           0.0.0.0:*
udp        0      0 0.0.0.0:55180           0.0.0.0:*
udp        0      0 0.0.0.0:60866           0.0.0.0:*
udp        0      0 0.0.0.0:45534           0.0.0.0:*
udp        0      0 0.0.0.0:60910           0.0.0.0:*
udp        0      0 0.0.0.0:52752           0.0.0.0:*
udp        0      0 0.0.0.0:36386           0.0.0.0:*
udp        0      0 0.0.0.0:53              0.0.0.0:*
udp        0      0 0.0.0.0:58425           0.0.0.0:*
udp        0      0 0.0.0.0:60999           0.0.0.0:*
udp        0      0 0.0.0.0:56926           0.0.0.0:*
udp        0      0 0.0.0.0:57958           0.0.0.0:*
udp        0      0 0.0.0.0:55409           0.0.0.0:*
udp        0      0 192.168.0.94:123        0.0.0.0:*
udp        0      0 127.0.0.1:123           0.0.0.0:*
udp        0      0 0.0.0.0:123             0.0.0.0:*
udp        0      0 0.0.0.0:54908           0.0.0.0:*
udp        0      0 0.0.0.0:25221           0.0.0.0:*
udp        0      0 0.0.0.0:55953           0.0.0.0:*
udp        0      0 0.0.0.0:32943           0.0.0.0:*
udp        0      0 0.0.0.0:38575           0.0.0.0:*
udp        0      0 0.0.0.0:46270           0.0.0.0:*
udp        0      0 0.0.0.0:5353            0.0.0.0:*
udp        0      0 0.0.0.0:48447           0.0.0.0:*
udp        0      0 0.0.0.0:36161           0.0.0.0:*
udp        0      0 0.0.0.0:33090           0.0.0.0:*
udp6       0      0 :::41920                :::*
udp6       0      0 :::53                   :::*
udp6       0      0 fe80::e43a:48ff:fe4:123 :::*
udp6       0      0 fdb9:f190:fc9b:0:e4:123 :::*
udp6       0      0 ::1:123                 :::*
udp6       0      0 :::123                  :::*
udp6       0      0 :::5353                 :::*
root@Milan-One:~#

nambiraja, 2020-12-17:
> No. Please find the below output of netstat.
>
> root@Milan-One:~# netstat --numeric-ports --udp --listening
> Active Internet connections (only servers)
> Proto Recv-Q Send-Q Local Address [...]

netstat output seems good I think. Please help to resolve.

skarg, 2020-12-18:
Can you show the network interfaces list:
`$ ifconfig`

skarg, 2020-12-18:
Wireshark or tcpdump can also be used to help figure out why the BACnet/IP requests are timing out.
List the network interfaces:
`tcpdump -D`
Monitor the BACnet/IP default UDP port on the specific interface:
`sudo tcpdump -n -i enp0s3 udp port 47808`
Monitor the BACnet/IP default UDP port on any interface:
`sudo tcpdump -n -i any udp port 47808`
In a separate terminal window, run the BACnet command line:
`./bin/bacwi -1`
See the results from tcpdump:
```tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked v1), capture size 262144 bytes
10:04:42.433527 IP 10.0.2.15.47808 > 10.0.2.255.47808: UDP, length 12
10:04:42.437490 IP 192.168.1.92.47808 > 10.0.2.15.47808: UDP, length 26
10:04:42.437845 IP 10.0.2.2.47808 > 10.0.2.15.47808: UDP, length 20
10:04:42.542434 IP 10.0.2.2.55067 > 10.0.2.15.47808: UDP, length 24
10:04:42.602166 IP 10.0.2.2.47592 > 10.0.2.15.47808: UDP, length 24
10:04:42.622178 IP 10.0.2.2.47592 > 10.0.2.15.47808: UDP, length 24
10:04:42.671485 IP 10.0.2.2.45548 > 10.0.2.15.47808: UDP, length 24
```

nambiraja, 2020-12-19:
Thanks for the reply. Will try and let you know.

nambiraja, 2020-12-19:
Please the below output:
root@Milan-One:/var/www/Milan/bacnet-stack-1.0.0/bin# sudo tcpdump -n -i eth0 udp port 47808
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
09:55:05.664104 IP 192.168.0.94.47808 > 192.168.0.255.47808: UDP, length 12
09:55:05.664787 IP 192.168.0.222.47808 > 255.255.255.255.47808: UDP, length 25

nambiraja, 2020-12-19:
Is there any problem my bacnet device broadcast iam message to 255.255.255.255.47808

Because I have another bacnet device it send the signal to 192.168.0.255.47808. 
My working device IP :192.168.0.68
Please the below working device signal:
root@Milan-One:/var/www/Milan/bacnet-stack-1.0.0/bin# sudo tcpdump -n -i eth0 udp port 47808
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
10:34:05.899165 IP 192.168.0.94.47808 > 192.168.0.255.47808: UDP, length 12
10:34:05.899847 IP 192.168.0.222.47808 > 255.255.255.255.47808: UDP, length 25
10:34:06.083484 IP 192.168.0.68.47808 > 192.168.0.94.47808: UDP, length 20
10:34:06.084577 IP 192.168.0.68.47808 > 192.168.0.94.47808: UDP, length 29

skarg, 2020-12-19:
The problem you are experiencing seems similar to a [question about broadcasting UDP packet to 255.255.255.255](https://stackoverflow.com/questions/19771485/broadcasting-udp-packet-to-255-255-255-255) at stack-overflow. 

Can you configure a default route in the route table?
`$ sudo route add default eth0`

nambiraja, 2020-12-21:
Thanks for your support. After changing my lan network (new router) now I got response from device. I will close this issue. And once again thank you very much Steve.

commit 63e4fa1f (2020-12-21): fixes #141 Linux BIP with no default gateway (#144)
closed 2020-12-21 by nambiraja