bacnet-stack/bacnet-stack #145: Bacnet MSTP

hyperiobacnet, 2020-12-22:
Hello, I'm working on io for bacnet.
bacnet ip successfully completed but I do not know about mstp.

1.Do I need to separate devices as master slaves ?  And how can I do this.?
2. Can you share with me sample code for mstp ?

skarg, 2020-12-22:
BACnet MS/TP is a datalink, so it is separate from device.  The datalink can be master node (peer to peer token ring, always token passing) or slave node (polled - like MODBUS).

BACnet MS/TP examples are numerous in the library.  The example apps in the apps/ folder can be built using MS/TP as the datalink (instead of BACnet/IP by default), there is a BACnet/IP to MS/TP router example in apps/ folder, there are numerous microcontroller MS/TP examples in ports/ folder.

hyperiobacnet, 2020-12-22:
thank u sir .
but I not see for bacnet slave example.
please send me link possible.
How do I separate as slave and master

skarg, 2020-12-22:
See [MSTP_Slave_Node_FSM](https://sourceforge.net/p/bacnet/src/ci/master/tree/src/bacnet/datalink/mstp.c#l1138)() function in mstp.c module, which can be seen in the [Linux port of dlmstp.c module](https://sourceforge.net/p/bacnet/src/ci/master/tree/ports/linux/dlmstp.c#l295), for example, when the MS/TP MAC address is above 127 (128-254).
```
            if (MSTP_Port.This_Station <= 127) {
                while (MSTP_Master_Node_FSM(&MSTP_Port)) {
                    /* do nothing while immediate transitioning */
                }
            } else if (MSTP_Port.This_Station < 255) {
                MSTP_Slave_Node_FSM(&MSTP_Port);
            }
```

hyperiobacnet, 2020-12-22:
thank u sir .

Is there a  mstp port for esp32

hyperiobacnet, 2020-12-22:
thank u sir

hyperiobacnet, 2020-12-24:
MSTP_Slave_Node_FSM () function is not available in some dlmstp files. 
my mac address = 150  does it still work as a slave ? 
my device esp32
Do I need to add the MSTP_Slave_Node_FSM function to make it work?

https://github.com/bacnet-stack/bacnet-stack/blob/ef1a3b2a7d0129032c5c050132e83f67450b5703/ports/stm32f10x/dlmstp.c#L1522

hyperiobacnet, 2020-12-26:
> See [MSTP_Slave_Node_FSM](https://sourceforge.net/p/bacnet/src/ci/master/tree/src/bacnet/datalink/mstp.c#l1138)() function in mstp.c module, which can be seen in the [Linux port of dlmstp.c [...]

MSTP_Slave_Node_FSM () function is not available in some dlmstp files.
my mac address = 150 does it still work as a slave ?
my device esp32
Do I need to add the MSTP_Slave_Node_FSM function to make it work?

skarg, 2025-02-06:
Yes, you would need to copy MSTP_Slave_Node_FSM() into any standalone dlmstp.c module to enable Salve mode support.  
Many of the standalone dlmstp.c modules have been refactored to use a common [src/bacnet/datalink/dlmstp.c](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/datalink/dlmstp.c).

closed 2025-02-06 by skarg