bacnet-stack/bacnet-stack #152: Cant build on FreeBSD 11

sparvu, 2021-02-08:
Is anyone trying even to run and build BACNET stack on FreeBSD 11/12 ? Under FreeBSD 11 64bit I cannot go far:

```
krmx@fbsd11:~/bacnet-stack-bacnet-stack-1.0.0 % gmake BACNET_PORT=bsd all
gmake -s -C apps all
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:435:16: warning: variable 'addr_ptr' is used uninitialized whenever 'if' condition is false [-Wsometimes-uninitialized]
    } else if (sockaddr_ptr->sa_family == AF_INET6) {
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:438:12: note: uninitialized use occurs here
    return addr_ptr;
           ^~~~~~~~
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:435:12: note: remove the 'if' if its condition is always true
    } else if (sockaddr_ptr->sa_family == AF_INET6) {
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:432:19: note: initialize the variable 'addr_ptr' to silence this warning
    void *addr_ptr;
                  ^
                   = NULL
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:465:24: warning: variable 'rv' is uninitialized when used here [-Wuninitialized]
                return rv;
                       ^~
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:451:12: note: initialize the variable 'rv' to silence this warning
    char rv; /* return value */
           ^
            = '\0'
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:471:24: warning: variable 'addr_ptr' is used uninitialized whenever 'if' condition is false [-Wsometimes-uninitialized]
            } else if (strcmp(request, "netmask") == 0) {
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:474:17: note: uninitialized use occurs here
            if (addr_ptr) {
                ^~~~~~~~
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:471:20: note: remove the 'if' if its condition is always true
            } else if (strcmp(request, "netmask") == 0) {
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:463:27: note: initialize the variable 'addr_ptr' to silence this warning
            void *addr_ptr;
                          ^
                           = NULL
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/bip-init.c:625:9: warning: unused variable 'sock_fd' [-Wunused-variable]
    int sock_fd = 0;
        ^
4 warnings generated.
/usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/mstimer-init.c:31:10: fatal error: 'mach/clock.h' file not found
#include <mach/clock.h>
         ^~~~~~~~~~~~~~
1 error generated.
gmake[2]: *** [Makefile:37: /usr/home/krmx/bacnet-stack-bacnet-stack-1.0.0/ports/bsd/mstimer-init.o] Error 1
gmake[1]: *** [Makefile:210: readprop.all] Error 2
gmake: *** [Makefile:34: apps] Error 2

```

Questions:

- what is this prefix ? bacnet-stack-bacnet-stack-1.0.0 ? can't you fix the release properly to have bacnet-stack only ?
- any ideas about FreeBSD 11 compile errors ?

skarg, 2021-06-12:
Thank you for posting the errors. I don't have FreeBSD in a VM to test the build, so I rely on others to submit patches and I try to be good at guessing what changes might affect the BSD build.

Can you elaborate on:

- what is this prefix? - I am not sure what is asked by this question. Can you be specific?
- "to have bacnet-stack only" - what specifically were you expecting?

skarg, 2021-06-12:
I downloaded a FreeBSD 11.4 VirtualBox VM, installed git, vim, gmake, and gcc.  I cloned bacnet-stack, and saw some build errors in ports/bsd datetime-init.c and mstimer-init.c files. I updated datetime-init.c and mstimer-init.c for ports/bsd to handle __MACH__ or else.  I added Makefile target for bsd to make it easy to build default BACnet/IP applications:
```
$ gmake bsd
$ export BACNET_IFACE=em0
$ ./bin/bacwi -1
```

closed 2021-06-13 by skarg