bacnet-stack/bacnet-stack #158: How to add multiple device object in BACnet server demo(using BACnet/IP).

ReshmaPatel24, 2021-02-24:
Setup details:
BACnet server application running on Linux system (Ubuntu 18.04).
Yabe(Yet another BACnet explorer) on windows system to get the BACnet the server information.
Bacnet Stack from branch [bacnet-stack-1.0](https://github.com/bacnet-stack/bacnet-stack/tree/bacnet-stack-1.0)

- To add multiple device object, we have modified Device_Count API so that we can have more than one device object. we can see multiple device objects in Yabe(yet another BACnet explorer) tool, but the fields(instance number, name) are same for all the device objects. 

Checked multiple demos but not able to find the one with multiple device object. The gateway demo is adding multiple routing device but the demo is not getting compiled(instruction used: make -s -C apps/gateway). Inside CMakeLists.txt gateway related lines(527 and 528) are commented.

The objective of my project is to have multiple BACnet device object and each device object should have multiple analog/binary input/output objects.

stevedh, 2021-03-17:
The standard specifies ... "There shall be exactly one Device object in each BACnet Device" (\S 12.11).  You have to have multiple device servers to achieve this, perhaps on different ports or virtual network segments.

ReshmaPatel24, 2021-03-18:
Ok, so is there any demo application available that creates multiple device object using virtual network segment or different ports?

skarg, 2021-03-19:
Ping shared his methods with me (2021), similar to how other users have used Docker to create thousands of BACnet devices for testing.  If someone wants to convert it into Markdown, I'm happy to add it to the source code doc folder.

[45554930-VirtualBACnetdevicesandBACnetOpenStack-230221-1949.pdf](https://github.com/bacnet-stack/bacnet-stack/files/6171686/45554930-VirtualBACnetdevicesandBACnetOpenStack-230221-1949.pdf)

Joseph shared this method with me (from 2016):
Only three files in the folder:
1. Precompiled copy of bacserv from some unknown version of the stack
2. Dockerfile to create a small distro and add the binary
3. bash script to spin up a bunch of containers

$ ls
```
bacserv  Dockerfile  start.sh
```

bacserv is compiled, from a now unknown version of the stack.

The docker file simply loads it into a small distro:

$ cat Dockerfile
```
FROM alpine:3.3
MAINTAINER JC
ADD bacserv /bin/bacserv
CMD ["/bin/bacserv", "1"]
```

$ cat start.sh
```
#!/bin/bash
set -e
for i in `seq 1 100`;
do
  let "bnPort=47000+$i"
  echo "Starting Container $i: $bnPort"
  sudo docker run -d  \
             -p $bnPort:47808 \
             bacserv /bin/bacserv $bnPort
done
``` 

Naveen reports: "Run docker on Linux instead of Window. The NAT of Docker for Windows isn't equipped with the ip translation I believe. On linux I can simply ping the running containers from the host but not on Windows. I can now read the bacnet demo server running on the host and a client running as a container."

ReshmaPatel24, 2021-03-22:
If we follow the above approach then do we need to open multiple ports in YABE to read write the specific device?

As per the objective of my project I'll be only listening to single port for all the devices.