bacnet-stack/bacnet-stack #161: Unable to build Gateway App under Debian Linux

nospamsam2011, 2021-03-19:
Good day All,

I clones the latest version of the bacnet-stack and did a make clean all with no real issues.  I tested the build via executing bacserv and all is working fine.  However, if I try and make the gateway demo (e.g. make gateway) I get the following errors:

root@som:~/bacnet-stack# make gateway
make -s -C apps gateway
main.o: In function `main':
main.c:(.text.startup.main+0x6a): undefined reference to `Routing_Device_Init'
collect2: error: ld returned 1 exit status
Makefile:67: recipe for target 'bacgateway' failed
make[2]: *** [bacgateway] Error 1
Makefile:214: recipe for target 'gateway' failed
make[1]: *** [gateway] Error 2
Makefile:66: recipe for target 'gateway' failed
make: *** [gateway] Error 2
root@som:~/bacnet-stack#

I have been reviewing the various makefiles and for some reason I cannot see any issues.  Am I missing something?   Mu O/S is a Debian based version of Linux on a SOM module.

Thanks in advance!

Cheers,

Sam

skarg, 2021-03-20:
Since the gateway example re-uses some of the same library files, but with BAC_ROUTING defined, the old files are still in the folder without the extra functionality needed by the gateway example. For gateway, start with the clean target for make.

    $ make clean
    $ make gateway

or under Windows (git bash or MinGW):

    $ make clean
    $ make gateway-win32

nospamsam2011, 2021-03-20:
Good day Steve,

Thank you very much for the info and guidance!  Your suggestion worked perfectly!
Thanks again!
Cheers,
Sam

closed 2021-03-20 by nospamsam2011