bacnet-stack/bacnet-stack #188: Compile faile when BBMD_ENABLED=0

MykytaDorokhin, 2021-08-07:
Hello,
Here are the linking errors I get when trying to build with BBMD_ENABLED=0.

```
bacnet-stack-1.0.0/src/bacnet/datalink/dlenv.c:159: undefined reference to `bvlc_register_with_bbmd'
bacnet-stack-1.0.0/src/bacnet/datalink/dlenv.c:214: undefined reference to `bvlc_bdt_list'
```

I'm not an expert on bacnet but still I need to develop a virtual device so to expose some values. As per my understanding, this device don't need to act as BBMD. Trying to set  BBMD_ENABLED=0 produces linking errors.

Does it mean all devices built with this library act as BBMD? Is it sort of a requirement or my ignorance?

BTW, is there a way to disable BBMD using cmake files? I havn't found it.

Thanks a lot,
Mykyta

MykytaDorokhin, 2021-08-13:
Sorry guys. Did I say something stupid? Please comment so I know where to dig. Thanks.

skarg, 2021-08-14:
Calls to those functions are probably missing "#if BBMD_ENABLED" around them.

skarg, 2021-08-14:
Note: To facilitate interoperability, BTL requires that a device support BBMD (4 entries) & FD registrations, or be able to register as FD in order that this BACnet device can be placed in any facility (site, building, etc) subnet and be able to be the only BACnet device in the subnet and still be able to communicate with other BACnet devices (workstation, building controller, etc).

MykytaDorokhin, 2021-08-15:
Thank you, Steve

MykytaDorokhin, 2021-09-24:
> Calls to those functions are probably missing "#if BBMD_ENABLED" around them.

Changing
#if defined(BACDL_BIP)
to
#if defined(BACDL_BIP) && defined(BBMD_CLIENT_ENABLED)
on top of dlenv_register_as_foreign_device function fixes the problem. 

Now it builds with
#define BBMD_ENABLED 0

closed 2025-02-06 by skarg