bacnet-stack/bacnet-stack #194: Problem with lwip

alikhm4528, 2021-09-01:
Hi
I'm just new to **bacnet** and i'm trying to use **bacnet** in **stm32f7** so i need to use **lwip** but i can't find implementation for **bip_receive** function in **bip.c** . I just searched for it in other versions but can't find anything. could you please help me to figure out what should i do ?

skarg, 2021-09-02:
The typical pattern is:
```
    /* handle the messaging */
    pdu_len = datalink_receive(&src, &PDUBuffer[0], sizeof(PDUBuffer), 0);
    if (pdu_len) {
        npdu_handler(&src, &PDUBuffer[0], pdu_len);
    }
```
The LwIP pattern uses a callback from the UDP port:
```
    /* Set a receive callback for the upcb */
    udp_recv(Server_upcb, bip_server_callback, NULL);
```
This callback then calls the npdu_handler():
```
void bip_server_callback() 
{
...
        if (pdu_len) {
            BIP_STATS_INC(recv);
            npdu_handler(&src, &pdu[pdu_offset], pdu_len);
        } else {
            BIP_STATS_INC(rxdrop);
        }

}
```
It's possible to add a queue that is fed in the callback, and read the queue via bip_recieve(), but I didn't choose to do it that way.

alikhm4528, 2021-09-05:
thank you very much steve

closed 2022-04-27 by skarg