bacnet-stack/bacnet-stack #205: CMake build does not support BACDL_ALL

AlexanderWells-diamond, 2021-11-11:
There seems to be no way to tell CMake configuration to set BACDL_ALL macro, only a series of individual BACDL_* macros e.g. BACDL_ETHERNET and BACDL_BIP. 

I'm trying to use this library to support both Ethernet and BIP communication at runtime (I'm actually upgrading a server from the old bacnet4linux library). This means I must compile a library with both the Ethernet and BIP macros. However, enabling both of these through CMake configuration results in a failing build:

```
[  0%] Building C object CMakeFiles/bacnet-stack.dir/src/bacnet/basic/npdu/h_routed_npdu.c.o
/usr/bin/cc -DBACDL_BIP -DBACDL_ETHERNET -DBACNET_PROPERTY_LISTS -DBACNET_PROTOCOL_REVISION=19 
-DBACNET_STACK_STATIC_DEFINE -DBAC_ROUTING -DPRINT_ENABLED=1 -I/<cut>/bacnet-src/src 
-I/<cut>/bacnet-src/ports/linux  -MD -MT CMakeFiles/bacnet-stack.dir/src/bacnet/basic/npdu/h_routed_npdu.c.o 
-MF CMakeFiles/bacnet-stack.dir/src/bacnet/basic/npdu/h_routed_npdu.c.o.d 
-o CMakeFiles/bacnet-stack.dir/src/bacnet/basic/npdu/h_routed_npdu.c.o 
-c /<cut>/bacnet-src/src/bacnet/basic/npdu/h_routed_npdu.c
/<cut>/bacnet-src/src/bacnet/basic/npdu/h_routed_npdu.c: In function ‘routed_apdu_handler’:
/<cut>/bacnet-src/src/bacnet/basic/npdu/h_routed_npdu.c:208:41: error: ‘BVLC_ORIGINAL_UNICAST_NPDU’ undeclared 
(first use in this function)
         if (bvlc_get_function_code() != BVLC_ORIGINAL_UNICAST_NPDU) {
                                         ^
```

The issue stems from `src/bacnet/datalink/datalink.h` and its cascading `#if defined ... #elif...` blocks starting at line 31. If Ethernet is specified then the BIP never gets to include `bvlc.h`, and hence that value remains undefined. I'll admit I don't obvious see how defining BACDL_ALL would solve this situation, as it looks like it would end up just not including `h_routed_npdu.c` line 208 and thus the functionality changes, but it would at least compile.

skarg, 2021-11-11:
There are a couple of example apps that support multiple datalinks at runtime using a router between them. Maybe their configuration would be helpful?  See router-ipv6 in the CMakeLists (router-mstp is missing from CMake build).
https://github.com/bacnet-stack/bacnet-stack/tree/master/apps/router-ipv6

The datalink.h will single home the device object and replies on one datalink (e.g. BACDL_BIP), and the router can route to and from the other datalink(s) as needed.

AlexanderWells-diamond, 2021-11-12:
Thank you for the information. It seems that the Makefile for that app doesn't work in the 1.0 tag, only on master. I also note that the CMake build for that app doesn't seem to work at all - if you define only BACDL_BIP6 many of the bip_ functions remain unresolved, and if you define both BIP and BIP6 several of the BIP6 data structures remain undefined. 

If I'm understanding the CMake file correctly, it appears to build the `libbacnet-stack.a` library file and link every app against that. This is the approach I want to take, I'm just confused because it doesn't seem possible to support both Ethernet and BIP - this router application only supports BIP and BIP6, and the MSTP router supports MSTP and BIP, but no Ethernet on either of them.

closed 2025-02-06 by skarg