bacnet-stack/bacnet-stack #214: Specifying BACNET_IP_PORT breaks example Server

AlexanderWells-diamond, 2022-01-19:
Specifying a custom BACNET_IP_PORT seems to break the example server - it no longer sends Bacnet packets to the network and is no longer visible using the WhoIs example.

I ran this on RedHat 8, and also specified the "lo" aka loopback adaptor to save trawling through a list of possible network adaptors the server might bind to.

To recreate:

- Build all applications
- `export BACNET_IP_PORT=6789`
- `export BACNET_IFACE=lo`
- `./bin/bacserv 1234`

And then in a separate terminal run the WhoIs program using the `bvlc.sh` helper file:
- `./bin/bvlc.sh 127.0.0.1`
- `./bin/bacwi -1`
- Observe no Devices are found

Simply unsetting the `BACNET_IP_PORT` variable and re-running these steps means the WhoIs program successfully finds the server.

I did observe in Wireshark that the server does send some packet, but Wireshark doesn't recognise it as part of the Bacnet protocol:
```
62	11.571837178	127.0.0.1	127.255.255.255	UDP	67	6789 → 6789 Len=25

0000   00 00 00 00 00 00 00 00 00 00 00 00 08 00 45 00   ..............E.
0010   00 35 61 1d 40 00 40 11 da 9a 7f 00 00 01 7f ff   .5a.@.@.Ú......ÿ
0020   ff ff 1a 85 1a 85 00 21 ff 32 81 0b 00 19 01 20   ÿÿ.....!ÿ2..... 
0030   ff ff 00 ff 10 00 c4 02 00 04 d2 22 05 c4 91 03   ÿÿ.ÿ..Ä...Ò".Ä..
0040   22 01 04                                          "..
```

I also tried a variety of experiments calling `bip_set_port(6789)` at various places in the demo server code but all of those had the same effect. (It looks to me like calling that function could easily cause problems, as the static BIP_Port variable is used to report back the Address in various functions and so could easily become out of line with the actual port the socket was started on, but that's probably an issue best left for another time)

I understand that somehow the WhoIs program is using port 47809, but that's also using BBMD which shouldn't be necessary for just running a single server. Or is there some part of the BacNet protocol that requires the 47808 port, and any other port must be registered as a Foreign Device?

skarg, 2022-01-19:
BACNET_IP_PORT is the UDP port used by the application (server, clients).  If the listener or sending is not using the same UDP port, then they will not be able to communicate.

Wireshark can 'see' an alternate (something other than 47808) UDP port for BACnet by expanding the UDP layer, selecting the Source Port, and using "Decode As..." and selecting BVLC as the decoder.
[image]

In the example above, export the same BACNET_IP_PORT for both terminal windows, or configure bvlc.sh to use the BBMD at UDP port 6789 instead of 47808.

skarg, 2022-01-19:
[Contemporary Controls](https://www.ccontrols.com/) has some very nice diagrams in their [router documentation](https://www.ccontrols.com/pdf/AGBASRTB00.pdf) to describe configuring a variety of BACnet/IP networks.

AlexanderWells-diamond, 2022-01-20:
Right, so it sounds like this isn't a bug, but me misunderstanding something. I don't quite follow why the WhoIs application cannot find the Server without being told where it's running. Isn't the point of a broadcast search to transmit to everything accessible (as this is running on localhost, I assume that should be 127.255.255.255), and then every port should receive the message and be able to respond to it?

And thank you for pointing out Wireshark's configuration menu for this - not sure I'd ever have realised that was the problem!

skarg, 2022-01-20:
Broadcast is limited to the broadcast subnet (not always 255.255.255.255), and every UDP port would be 0..65535 (64K of them!) - usually something done with [NMAP](https://nmap.org/book/scan-methods-udp-scan.html) and not with BACnet clients.

Typical BACnet/IP server devices use UDP port 47808 (0xBAC0) by default, and are configurable on sites so that they could use something else (0xBAC1, 0xBAC2, etc) - but normally configured the same for every device on site - or at least groups of devices on site.  Some sites split floors or wings or buildings to use a single unique UDP port and access each of them from a building controller or workstation that can have multiple client connections.

AlexanderWells-diamond, 2022-01-21:
Right, thank you for the information :)

closed 2022-01-21 by AlexanderWells-diamond