bacnet-stack/bacnet-stack #224: BBMD and Docker

AlexMAS, 2022-02-14:
Hello!

First of all, thanks for your project!

I have a question about using BBMD functionality.

Currently to test BACnet I run `backnet-stack` in Docker and it works fine.

**docker-compose.yml**
```yml
ersion: '3.7'

services:

  # Defines the virtual BACnet device
  bacnet_device:
    build:
      context: ./
      dockerfile: ./Dockerfile
    image: bacnet
    container_name: bacnet_device
    command: bacserv 200001
    networks:
      - dev
```

**Dockerfile**
```Dockerfile
FROM ubuntu:20.04 AS build

RUN apt-get update && apt-get -y install build-essential

WORKDIR /bacnet-stack
COPY bacnet-stack ./
RUN make

FROM ubuntu:20.04 AS runtime

COPY --from=build /bacnet-stack/bin/* /usr/local/bin/
ADD bacnet-wrapper /

RUN chmod a+x /bacnet-wrapper

EXPOSE 47808:47808/udp

CMD ["/bacnet-wrapper"]
```

**bacnet-wrapper**
```bash
#!/bin/bash

bin=/usr/local/bin

# if ($1 != "" && fileExists($bin/$1) && fileExecutable($bin/$1))
if [[ x"$1" != x && -f $bin/$1 && -x $bin/$1 ]]; then
    exec $@
else
    echo 'Usage: docker run -it bacnet <command> <args>'
    echo '... where <command> is one of:'
    ls -C $bin
    exit 1
fi
```

To development we just added this container to our project network.

But in production environment we have to work with BACnet devices by using BBMD. And now we would like to setup a test environment like we have to check that our app works fine. I suppose that we need to do the next steps.

* Run `bacserv` in one Docker container (like above mentioned) in one network (let's say `net1`)
* Run a BBMD gateway in the app network (that's the question - I don't know how to do it)
* Register the virtual device from `net1` with the BBMD gateway (also the question)

I'm not so familiar with BACnet and to be honest understand only base principles of the protocol :)

Thanks!

AlexanderWells-diamond, 2022-02-14:
I'm not an expert on this, but there's a little bit of support built in for BBMD. Have you seen the "bvlc.sh/.bat" scripts? There's a brief user guide for them in `doc/README.faq` question 15. That script is how I run my local testing, but unfortunately I don't know if/how that'll run in Docker.

AlexMAS, 2022-02-14:
@AlexanderWells-diamond Thanks for reply. As I understand `bvlc.bat` configures environment variables (`BACNET_BBMD_ADDRESS` and `BACNET_IP_PORT`) for any subsequent `bacnet-stack` tools. I'm trying to figure out how these variables are used. Can one of `bacnet-stack` tool work as a virtual BBMD? Or by using these variables I can only point to an existing BBMD gateway.

skarg, 2022-02-14:
bacserv (apps/server) demo incorporates a BBMD device that:
a) has a BDT (broadcast device table) that could be filled with peer subnets that allow broadcasts to propagate. The BDT can be filled via environment variable values or via Write BDT client application
b) has a FDT (foreign device table) that accepts foreign device registration from clients

bacXX (apps/whois, apps/readprop, apps/writeprop, etc) demos incorporate a BBMD client device that:
a) registers to a BBMD as a foreign device
b) re-registers as needed before subscription expires
c) sends a broadcast using bvlc_send_pdu() which check Remote_BBMD.port for non-zero.  The non-zero Remote_BBMD.port indicates that they are a foreign device, and the broadcast is sent as "distribute broadcast to network" instead of "original broadcast" frame type (for non-foreign-devices).

skarg, 2022-02-14:
The [datalink/dlenv.c](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/datalink/dlenv.c) module manages the environment variables for the BBMD used in the demo apps.  Each BDT entry can be set like:
`BACNET_BDT_ADDR_1=192.168.1.1`
note that BACNET_BDT_ADDR_0 is reserved and configured for the device itself.

AlexMAS, 2022-02-15:
@skarg Thank you for so comprehensive answer. The issue can be closed.

closed 2022-02-15 by AlexMAS