bacnet-stack/bacnet-stack #256: Some questions about MSTP and IP protocol.

erhnam, 2022-04-26:
Good morning,

First of all I would like to congratulate you for the great work you are doing with this library. Before using this library for my project I would like to ask some questions to check compatibility.

Is it possible to use several MSTP devices?
Is it possible to use several IP devices? 
Is it possible to use MSTP and IP at the same time?
Is this library compatible with threads?

Is it possible to use several USB ports as MSTP at the same time? My hardware has four RS485 ports and I would like to distribute the devices between the ports.

Regards.

BACnetEd, 2022-04-26:
Erhnam

I have successfully used multiple datalinks, multiple devices, MS/TP & IP
simultaneously and linux threads in multiple projects using a fork of this
stack.

The fork can be found at https://github.com/BACnetTesting/BACnet-Stack .
There are multiple branches as follows (names off the top of my head):

   - Incoming - an exact copy of Steve Karg's latest, merged to
   - MSVC - a groomed subset of the above, compilable in Windows Visual
   Studio, merged to
   - Feature creep - I have started adding 'additional features here',
   merged to
   - Multiple Datalinks - this allows a single executable to run different,
   simultaneous datalinks per your one request, but is not all that useful in
   practice, merges to
   - Full Routing - Skip this one unless all you want to do is create a
   BACnet router. Note that Steve's stack now provides a routing layer, but I
   have never used that. This branch merges to
   - Virtual Devices - Provides full routing AND support for multiple
   virtual devices per , very useful when mapping 3rd party devices onto
   BACnet, per "H.1.1.2 Multiple "Virtual" BACnet Devices in a Single Physical
   Device" of the BACnet specification

Caveats: This is an _old_ copy, but it is the foundation for at least 3 BTL
approved products. If you are interested, I  could freshen this repo up
with my latest WIP, and provide a working "Virtual Devices" demo, but from
that point on all I can do is provide 'guidance' because I am rather busy.

Let me know.
Ed

erhnam, 2022-04-26:
> 

Ed,

I'm going to have a look at your fork and if I have any questions I'll let you know. I hope I'm not bothering you too much.

Thank you very much Ed, really.

BACnetEd, 2022-04-26:
No worries. I have just updated this repo from 2019, and trimmed obsolete
devices and made the VirtualDevices the default branch. Ignore the others.
The VirtualDevices branch was recently compiling, but I have no idea if the
copy in the repo will compile. However, it is suitable for you to take a
good look at and decide. Then if you want to proceed, I can get it
compiling and routing IP-to-IP on either windows or linux, (lmk win or
linux) and then you should be all set.

erhnam, 2022-04-26:
Hi Ed,

Your library is giving me a lot of errors compiling with either make or cmake. (BACnet-Stack/src/eLib/ipc/cJSON/cJSON.c:26:21: fatal error: osLayer.h: No existe el fichero o el directorio #include "osLayer.h"
)

Does the original repository also allow all my requirements?

Thank you.

BACnetEd, 2022-10-11:
->  Does the original repository also allow all my requirements?

Do you mean "Does the Karg stack meet all your requirements?" - At this
point probably. There is a new routing layer, but as I said, I have not
tested that. I know that linux threads were used in a prior attempt at a
routing layer (which was by no means complete). So threads can be made to
work. I don't know if the routing layer allows mixed datalinks. You will
have to ask others... I am not sure if outgoing MS/TP packets will
be queued. Ditto. (ask others).

If you wish to pursue my fork, let's take that conversation offline,
contact me at ***@***.***

Ed