bacnet-stack/bacnet-stack #278: Schedule read/write support

MightyPork, 2022-05-23:
We're trying to interface a Siemens PLC and one of the things needed is reading and writing time programs using the Schedule object.

This is not currently supported and results in "Malformed ACK" error when read using RPM.

I'll work on trying to hack support into our copy of the library, not sure if it will be of sufficient quality for a pull request. 

Please let me know if there's some plans for this or ideas how to do it best. My current idea is adding it to the union in the `BACnet_Application_Data_Value` struct, but the schedules struct is rather big and might need dynamic allocation and cleanup.

packet capture file for wireshark (replace extension with .pcap, github refused to upload it as .pcap)
[trace-rpm-schedule.pcap.txt](https://github.com/bacnet-stack/bacnet-stack/files/8754866/trace-rpm-schedule.pcap.txt)

MightyPork, 2022-05-24:
Thanks, that looks exactly like what I want! I'll see if the part about extra types can be ported to current master

MightyPork, 2022-06-28:
Hi Steve, just to follow up on this, I see the pull request with snprintf is merged. Is the library now in a good state to try porting the schedule parsing (and other complex objects) from "jbennet"?

skarg, 2022-07-01:
Nearly! 

I have been adding more fixes to bacapp.c and bacapp.h to handle the Color Object data types:
https://github.com/bacnet-stack/bacnet-stack/compare/master...feature/color-objects-color-command

I added bacapp_decode_known_property() which is needed in rp_ack_print_data() to replace bacapp_decode_application_data().

skarg, 2022-07-14:
For your branch, can you put the schedule extra data types into dailyschedule.c module and dailyschedule.h header, rather than in bacdcode.c and bacothertypes.h?  bacothertypes grinds on my sensibilities like using temp or tmp or other such names, and bacdcode.c is already gigantic.

I'd also prefer to use function names that are Linnaean (start with the big and work towards the small, usually end with the action), such as
- datetime_application_decode()
- datetime_context_decode()
- datetime_application_encode()
- datetime_context_encode()

I revised the encode/decode library so that the encode allows for NULL buffer and returns the length in that case.  
The decode always passes in the size of the buffer and checks for size enough to decode.

Unit tests are under test/ folder, and follow the same folder hierarchy as src/ folder.  For the tests I usually start by copying an existing test folder that is similar in test style, and then edit the tests for the specific module.  If you want to write the unit tests. let me know.

MightyPork, 2022-07-14:
Okay, thanks for the info and guidance. I somehow missed the previous reply.

I'm planning to work on this in august, I'll be away for a week and then there's other work to finish first. I'll get back to you if I hit any problems. I will redo the porting from scratch on top of your latest master, it'll be easier than trying to rebase the experimental branch in my fork.

skarg, 2022-07-15:
Just a note: I already pushed the [color and color temperature objects and their properties](https://github.com/bacnet-stack/bacnet-stack/commit/38d213b47c87725abdad7f198b98703bf64b2f41) into master branch.

MightyPork, 2022-08-02:
Hi Steve, I have a quick question, now looking into adding the schedule support:

There are many functions like `int bacapp_decode_time_value(uint8_t *apdu, BACNET_TIME_VALUE *value)`, that do not have the `max_apdu_len` parameter. This can segfault if the apdu buffer is too short.

Some functions, like the new `xy_color_decode()` are written safely.

This can be seen in `bacapp_decode_known_property()` where some properties are decoded with length checking, some aren't.

Should I change `bacapp_decode_time_value()` and other functions I encounter to use `max_apdu_len` (breaks ABI if anyone uses that), or add e.g. `bacapp_decode_time_value_safe()` and use that? 

The old jbennet branch uses the safe version. I'm trying to reuse code from there, so I'd rather not remove the length checking.

MightyPork, 2022-08-02:
my WIP is now in a pull request #319 (not ready for merging yet)

skarg, 2022-08-02:
In an effort to avoid breaking the API before a major version number change, I've been (occasionally) copying and modifying bacapp_decode_xyz() or decode_xyz() and rewriting them as bacnet_xyz_decode() which adds the max_apdu_len argument without changing the API (i.e. new name has new API) - instead of using the name pattern with _safe() appended. Then I change the old function to call the new function, and someday will deprecate and remove the old functions. For example:
```
int decode_octet_string(
    uint8_t *apdu, uint32_t len_value, BACNET_OCTET_STRING *value)
{
    const uint16_t apdu_len_max = MAX_APDU;

    return bacnet_octet_string_decode(apdu, apdu_len_max, len_value, value);
}
```

Some of the new decode functions do add the max_apdu_len but don't follow bacnet_xyz_decode() naming pattern (a task for someone in the future, I suppose).

Since bacapp_decode_time_value() is existing function, using bacapp_decode_time_value_safe() should be okay, but you could also use the bacnet_time_value_decode() name pattern.

MightyPork, 2022-08-03:
Schedule decoding is now working, I tested it with real hardware:

```
$ ./bin/bacrp 2098178 17 10 123
(Mon: [02:00:00.00 inactive, 07:35:00.00 active]; Tue: [02:00:00.00 inactive, 07:35:00.00 active]; Wed: [02:00:00.00 inactive, 07:35:00.00 active]; Thu: [02:00:00.00 inactive, 07:35:00.00 active]; Fri: [02:00:00.00 inactive, 07:35:00.00 active]; Sat: [02:00:00.00 inactive, 07:35:00.00 active]; Sun: [02:00:00.00 inactive, 07:35:00.00 active])

$ ./bin/bacrp 2098178 17 10 123 1
{([02:00:00.00 inactive, 07:35:00.00 active])}

$ ./bin/bacrp 2098181 17 1 123
(Mon: [00:00:00.00 3, 02:00:00.00 2, 07:00:00.00 3]; Tue: [00:00:00.00 3, 02:00:00.00 2, 07:00:00.00 3]; Wed: [00:00:00.00 3, 02:00:00.00 2, 07:00:00.00 3]; Thu: [00:00:00.00 3, 02:00:00.00 2, 07:00:00.00 3]; Fri: [00:00:00.00 3, 02:00:00.00 2, 07:00:00.00 3]; Sat: [00:00:00.00 3, 02:00:00.00 2, 07:00:00.00 3]; Sun: [00:00:00.00 3, 02:00:00.00 2, 07:00:00.00 3])
```

The CI fails with the windows build, everything else is OK. I also added a simple unit test.

Encoding is not yet tested beside the unit test.

skarg, 2022-08-03:
Boy, that Windows build is noisy (warnings!). I'll add that to the to-do list.
```
bactimevalue.c
D:\a\bacnet-stack\bacnet-stack\src\bacnet\bactimevalue.c(181,54): error C2059: syntax error: '}' [D:\a\bacnet-stack\build\bacnet-stack.vcxproj]
```

MightyPork, 2022-08-04:
Thanks, what a weird restriction. It's fixed. I'll now try to test the encoding part

skarg, 2022-08-05:
re: platform abstraction (src/bacnet/bacnet_plat_compat.h)

I ran into an issue some months ago with platforms that don't support some libc string functions and proposed something similar to OpenThread in this issue: https://github.com/bacnet-stack/bacnet-stack/issues/204

mbedtls [platform.h](https://github.com/Mbed-TLS/mbedtls/blob/development/include/mbedtls/platform.h) file starts with:
>  The platform abstraction layer removes the need for the library to directly link to standard C library functions or operating system services, making the library easier to port and embed. Application developers and users of the library can provide their own implementations of these functions, or implementations specific to their platform, which can be statically linked to the library or dynamically configured at runtime.

soap solves it like this:
https://github.com/alisw/gsoap/blob/master/gsoap/stdsoap2.h

I prefer the mbedtls pattern, so maybe the name is src/bacnet/platform.h and added as `#include "bacnet/platform.h"` in the bacdef.h file?  We could also refactor the other platform specific code into it (search for _MSC_VER for example) and place the library implementations under bacnet/basic/sys/ as suggested in the other issue or keep it simple with src/bacnet/platform.c file in the same [mbedtls platform.c](https://github.com/Mbed-TLS/mbedtls/blob/development/library/platform.c) pattern.

MightyPork, 2022-08-08:
Okay, for now I moved the platform file to `src/bacnet/platform.h`, other refactoring / cleaning can be done later.
I've now tested `bacwp` with schedule and it worked! So I think the PR is ready for some final cleaning and merging.

Arm build is failing now. Does the weeklyschedule file need to be added to some special makefile?

About schedule printinf/parsing:
- The syntax in snprintf/parse is `(TAG; Weekeday: [Time value, ...], ...)` with `Weekday:` and all whitespace optional.
- When only a single day is read or written, the format is `(TAG; Weekeday: [Time value, ...])` (+ the array index 1-7 must be specified)
- TAG can be the bactext name, or a numeric value, e.g. 1=Boolean, 9=Enumerated. It must be specified explicitly as part of the text format
- Since array index is not passed to the encode and decode functions, I had to add a `singleDay` field to the WeeklySchedule struct. This is because when a single day is read or written using array index, the encoding is different (it's sent as a "single-day week").

```c
    typedef struct BACnet_Weekly_Schedule {
        BACNET_DAILY_SCHEDULE weeklySchedule[7];
        bool singleDay;
    } BACNET_WEEKLY_SCHEDULE;
```

One odd thing is that enumeration is transformed to "inactive" / "active" when printed, but it can't be parsed from that (the parser uses `strtoul()`). I think boolean does the same thing.

```
$ ./bin/bacrp 2098178 17 2 123
(Enumerated; Mon: [07:20:00.00 active, 22:50:00.00 inactive]; Tue: [07:20:00.00 active, 22:50:00.00 inactive]; Wed: [07:20:00.00 active, 22:50:00.00 inactive]; Thu: [07:20:00.00 active, 22:50:00.00 inactive]; Fri: [07:20:00.00 active, 22:50:00.00 inactive]; Sat: [07:20:00.00 active, 22:50:00.00 inactive]; Sun: [07:20:00.00 active, 22:50:00.00 inactive])

$ ./bin/bacwp 2098178 17 2 123 8 -1 30 '(Enumerated; Mon: [07:19:00.00 1, 22:50:00.00 0]; Tue: [07:20:00.00 1, 22:50:00.00 0]; Wed: [07:20:30.00 1, 22:50:00.00 0]; Thu: [07:20:00.00 1, 22:50:00.00 0]; Fri: [07:20:00.00 1, 22:50:00.00 0]; Sat: [07:20:00.00 1, 22:50:00.00 0]; Sun: [07:20:00.00 1, 22:50:00.00 0])'
```

Some other changes:
- I disabled the check in bacwp and bacwpm that limited tag to 0-15, I needed 30 to test schedule
- `bacapp_parse_application_data` now checks errors when using `strtol`, `strtoul`, `strtod`
- The old unsafe functions for TimeValue are now marked as deprecated (e.g. `bacapp_encode_time_value`)

I'm not sure about naming, so please change it as needed - e.g. `bacnet_time_value_context_decode` vs `bacnet_time_value_decode_context`, or should it be `bacapp_...`? is there a distinction?

Also, should I try to clean the commits somehow or it's okay as is?

MightyPork, 2022-08-08:
Okay the arm builds are now all fixed, I had to edit makefiles in the ports folder.

skarg, 2022-08-08:
>  I needed 30 to test schedule

Since those complex application tags are "whatever is defined in the stack" and not BACnet standard values, please use "-1" and have the WriteProperty application(s) "parse a known property" to determine the value and tag.

If we refactor bacapp_decode_known_property() into bacapp_known_property_tag() to return the internal tag (or -1 if not found):
```
        ...
        property_tag = strtol(argv[tag_value_arg], NULL, 0);
        args_remaining--;
        if (args_remaining <= 0) {
            fprintf(stderr, "Error: not enough tag-value pairs\r\n");
            return 1;
        }
        value_string = argv[tag_value_arg + 1];
        args_remaining--;
        if (property_tag < 0) {
            property_tag = bacapp_known_property_tag(Target_Object_Property);
        } else if (property_tag >= MAX_BACNET_APPLICATION_TAG) {
            fprintf(stderr, "Error: tag=%ld - it must be less than %u\r\n",
                property_tag, MAX_BACNET_APPLICATION_TAG);
            return 1;
        }
        if (property_tag >= 0) {
            status =
                bacapp_parse_application_data(property_tag, value_string,
                &Target_Object_Property_Value[i]);
            if (!status) {
                /* FIXME: show the expected entry format for the tag */
                fprintf(stderr, "Error: unable to parse the tag value\r\n");
                return 1;
            }
        } else {
            /* FIXME: show the expected entry format for the tag */
            fprintf(stderr, "Error: unable to parse the known property"
            " \"%s\"\r\n", value_string);
            return 1;
        }
        Target_Object_Property_Value[i].next = NULL;
        ...
```

MightyPork, 2022-08-18:
ok, it's now changed following your example and the <16 check added back

skarg, 2022-09-05:
Thank you for the contribution!

closed 2022-09-05 by skarg