bacnet-stack/bacnet-stack #279: minor improvements

MightyPork, 2022-05-25:
I have put my fork to github and rebased it on top of master. These are changes I needed when integrating the library into a larger project.

https://github.com/MightyPork/bacnet-stack/commits/fbx

Steve @skarg, could you look at the changes and say which ones you want? I'll make PRs.

Not sure about that `build.sh` file, I had to revert 1c1b6762472d89851ebf7d9bd15e99d21c8f5de6 as it stopped working on Linux. It should probably be split to two files, one for real Linux and one for MinGW.

The "[single property COV](https://github.com/MightyPork/bacnet-stack/commit/c66af266463af16c84d357dff099c0f47f8f7bf5)" thing - I tried it and it doesn't work (I still get COV for other properties on the object). I'm not sure if it's a bug in the bacnet server, or I did something wrong.

AlexanderWells-diamond, 2022-05-25:
I'm interested in the addition of the `debug_disable()` functions. Would it also be possible/practical to include one for the `BIP_DL_Debug` variable in `dlenv.c`? It's switched on via the `"BACNET_IP_DEBUG"` [env variable here](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/datalink/dlenv.c#L476), along with the BIP and BVLC debug. 

Depending how thoroughly you wish to do that, there's also the bip6 and bvlc6 debug flags, which follow the same pattern.

skarg, 2022-05-25:
[build.sh to use all CPU cores and remove wrong windows aliases](https://github.com/MightyPork/bacnet-stack/commit/d1d70aa7efa0a64f4744996a8b9d8523da3f2046)
 - I changed build.sh back (ref 0.8 branch) to mirror build.bat which uses MinGW (windows) and the Makefile, rather than using cmake as the builder.  The cmake native build can be invoked using `make cmake` since there is a recipe in the Makefile for it.  If build.sh is commonly used for cmake invocation, please PR.

[Allow "address_mac_from_ascii()" to parse hex 1-6 bytes](https://github.com/MightyPork/bacnet-stack/commit/b3c3ad4b034e3ca36986fcdc3d32c812556958e1) - please PR.

[Add option to subscribe to single property COV](https://github.com/MightyPork/bacnet-stack/commit/c66af266463af16c84d357dff099c0f47f8f7bf5) - please PR

[Make "dlenv_network_port_init()" public (was static)](https://github.com/MightyPork/bacnet-stack/commit/7c7f3e34c46ebf98dd4e963802517dcc8c80d94b)  - please PR

[Add "bvlc_debug_disable()"](https://github.com/MightyPork/bacnet-stack/commit/0db11fd5f42c56aed5e40f1af03fded09432b470) - please PR

 
[Add "bip_debug_disable()"](https://github.com/MightyPork/bacnet-stack/commit/462680657925642987948e364863328a92ba9dda) - please PR

[Lower minimal cmake version requirement](https://github.com/MightyPork/bacnet-stack/commit/dce0c8c472e7820873c976047f48647453b6dedf) - please PR

MightyPork, 2022-05-25:
I've split the changes to PRs, let's continue in the individual PRs.

closed 2022-05-25 by MightyPork