bacnet-stack/bacnet-stack #286: MyErrorHandler not working properly with WPM

philippejohnrichard, 2022-05-26:
The MyErrorHandler gives me the error code  ERROR_CODE_CONFIGURATION_IN_PROGRESS instead of ERROR_CODE_WRITE_ACCESS_DENIED when one of the property of WPM is not writable.

I need to look at Wireshark every time that I want to know which property gave me the error. 

1- How can I trace which :  Device, ObjectID, ObjectInstance and Property that caused the error.

2-How can I handle the error message frame to retrace the proper ErrorCode and ErrorClass in this specific use case where MyErrorHandler does not fully work. 

Thank you for the help, this stack is great btw. Good job !

philippejohnrichard, 2022-05-31:
I manage to quick fix my handler by adding a function in the apdu_handler()  {

        printf("SERVICE CHOICE : %u\n",service_choice);
        
             if(service_choice== SERVICE_CONFIRMED_WRITE_PROP_MULTIPLE){
                 my_function_to_handle_special_WPM(&apdu[0],len,src);
             }
             
 }
 
my_function_to_handle_special_WPM reads 1 by 1 the bytes form apdu[index]  and decode the received err message. 

Passing this information to my error handler by a static variable.

skarg, 2022-06-02:
Thank you for the solution and finding the problem!

In h_apdu.c module apdu_handler() function, my naive self created the error callback API assuming that the only data was the error code and error class, not realizing that additional data could be appended to that service. Perhaps an 'extended' error callback service should be added for the various services that need them, or change the API for all to provide the entire service data.

closed 2022-06-03 by skarg