bacnet-stack/bacnet-stack #298: Specify baudrate 76800 with port dev/ttySC0

philippejohnrichard, 2022-06-18:
I'm having problem trying to read/write to device while using the speed 76800 with the port dev/ttySC0.

I receive data, but always wrong frame. (No header frame 0x55) 

Is there a way to configure the _termios.h_ flags or maybe the the :  _serial_struct_ RS485_oldserial  in order to be able to communicate at that speed?

Note : 
-There is no problem with the port  dev/ttySC0 when communicating at 38400.
-There is no problem when the port  dev/ttyUSB0 when  communicating at 76800.

Thanks alot for your help

philippejohnrichard, 2022-06-18:
Additional info :   I currently run these test on a raspberry Pi 4 model B.

skarg, 2022-06-20:
The mechanism this library uses for Linux serial port 76800 bps is a [custom divisor](https://stackoverflow.com/questions/3192478/specifying-non-standard-baud-rate-for-ftdi-virtual-serial-port-under-linux) (76800 bps falls into the category of a non-standard baud rate for PC (personal  computer) based communications). It's possible that the Raspberry Pi 4 model B has a serial port that does not support the custom baud rate(s).

MightyPork, 2022-08-02:
I didn't test with RPi4, but I know from experience that RPi3 had this limitation. Using `BOTHER` didn't work, even when used with one of the standard baud rates.

ryanjmulder, 2025-05-29:
Hi @philippejohnrichard, we just committed #1007, which fixes a similar problem for me, and I think it'll fix yours, too.
Could you try it and find out?