bacnet-stack/bacnet-stack #301: Option to pass the serial device while using MSTP

raghavan97, 2022-07-12:
We can build the apps using the MSTP as Datalink layer (by doing make mstp).

This ties the device /dev/ttyUSB0 based on the default value of static char *RS485_Port_Name in rs485.c

When I want to do a who is , I would typically change the above variable to "/dev/ttyUSB0", build the binary and run ./bacwi application.

If I need to change my serial port , I would again have to recompile the binary after changing the *RS485_Port_Name in rs485.c 

It would be nice, if we can pass the serial port device (e.g. /dev/ttyUSB0, or /dev/ttyUSB1 and so on ) , when we invoke the apps like ./bacwi etc when the Data link is MSTP.

The mstpcap already has this facility.

skarg, 2022-07-14:
BACnet MSTP for the apps uses environment variables for the various options, rather than clutter the command line.  See bacnet/datalink/dlenv.c module for specifics:
```
 * - (Any):
 *   - BACNET_APDU_TIMEOUT - set this value in milliseconds to change
 *     the APDU timeout.  APDU Timeout is how much time a client
 *     waits for a response from a BACnet device.
 *   - BACNET_APDU_RETRIES - indicate the maximum number of times that
 *     an APDU shall be retransmitted.
 *   - BACNET_IFACE - set this value to dotted IP address (Windows) of
 *     the interface (see ipconfig command on Windows) for which you
 *     want to bind.  On Linux, set this to the /dev interface
 *     (i.e. eth0, arc0).  Default is eth0 on Linux, and the default
 *     interface on Windows.  Hence, if there is only a single network
 *     interface on Windows, the applications will choose it, and this
 *     setting will not be needed.
...
 * - BACDL_MSTP: (BACnet MS/TP)
 *   - BACNET_MAX_INFO_FRAMES
 *   - BACNET_MAX_MASTER
 *   - BACNET_MSTP_BAUD
 *   - BACNET_MSTP_MAC
```
For MSTP, use BACNET_IFACE=X as the environment variable.

For linux, use /dev/ttyUSB0 or whichever serial interface you are using. 
For Windows, use COM1 or whichever com port you are using.

datalink_init() is called with BACNET_IFACE as the string, which calls dlmstp_init() for MSTP, which calls RS485_Set_Interface() with the name.

For Linux, you can pass the environment variable like (or put it into a shell script, etc):
```
$ export BACNET_IFACE=/dev/ttyUSB1 ; /bin/bacwi 
```

raghavan97, 2022-08-04:
Steve,

Thanks a lot for your response. I tested it out today in my lab setup. It works well.

admin@viconics-test:~$ export BACNET_IFACE=/dev/ttyUSB0;sudo ./bacwi 1 20 --retry 5
RS485: Initializing /dev/ttyUSB0 at Baud Rate 38400=success!
MS/TP MAC: 7F
MS/TP Max_Master: 7F
MS/TP Max_Info_Frames: 1
;Device   MAC (hex)            SNET  SADR (hex)           APDU
;-------- -------------------- ----- -------------------- ----
  1       01                   0     00                   480
  2       02                   0     00                   480
  3       03                   0     00                   480
  4       04                   0     00                   480
;
; Total Devices: 4

closed 2022-08-04 by raghavan97