bacnet-stack/bacnet-stack #322: Help requested on a MSTP issue

raghavan97, 2022-08-04:
Team,

I am facing a strange situation with a MSTP network.

I am looking for debugging ideas using any of the tools like mstpcap or others.

Here is the scenario.

At the start , we have about 5 devices on the MSTP network and all of them are communicating well. I can see the devices using mstpcap and also by bacwi.

After sometime, none of the devices are talking on the network.

In that state mstpcap returns 

admin@FL103692-BE:~$ sudo ./mstpcap --baud 38400 /dev/ttyUSB0
RS485: Initializing /dev/ttyUSB0 at Baud Rate 38400=success!
mstpcap: Using /dev/ttyUSB0 for capture at 38400 bps.
mstpcap: saving capture to mstp_20220804120559.cap
2000 packets, 0 invalid frames
2400 packets, 0 invalid frameses
==== MS/TP Frame Counts ====
MAC     Device  Tokens  PFM     RPFM    DER     Postpd  DNER    TestReq TestRsp
4       -       0       2338    0       0       0       0       0       0
Node Count: 1

==== MS/TP Usage and Timing Maximums ====
MAC     MaxMstr Retries Npoll   Self/TT Treply  Tusage  Trpfm   Tder    Tpostpd
4       127     0       0       0       0       59      0       0       0
Node Count: 1
Invalid Frame Count: 0

Evidently there are no tokens that are passing.

For understanding the issue better, I added a few counters in the FSM  and took a snap shot of the counters in a space of around 60 secs approximately. 

At t1:
name=/dev/ttyUSB0 token_count=0 recv_errors=2 next_station=5 line_active=1 invalid_frames=0 rs485_xmit=448 rs485_recv=58696 pfm_send=0 pfm_reply=56

At t1+60 secs (approx)
name=/dev/ttyUSB0 token_count=0 recv_errors=4 next_station=7 line_active=1 invalid_frames=0 rs485_xmit=1352 rs485_recv=176045 pfm_send=2 pfm_reply=166

We can see that there is some traffic on both rs485 send and recv , but token count is not going up.

I dont have a person to help me onsite right now. But he did tell me that we had to reboot all the BACnet devices on the network and then things would be fine for sometime before which we get into this issue.

We have this problem in a couple of sites. 

I have a site where the only device on the MSTP network is the meter - https://shop.dentinstruments.com/products/powerscout-3-hd-power-submeter. I am seeing the issue there too.

Does anyone have an experience with that device ?

I am running on this hardware - https://www.advantech.com/products/9a0cc561-8fc2-4e22-969c-9df90a3952b5/uno-420/mod_2d6a546b-39e3-4bc4-bbf4-ac89e6b7667c

I will be happy to share any data that can be gleaned by running any of the bacnet-stack binaries.

Thanks a lot folks.

Bye,
Raghavan V

raghavan97, 2022-08-04:
The pcap file 
[mstp_20220804120559.cap.zip](https://github.com/bacnet-stack/bacnet-stack/files/9261815/mstp_20220804120559.cap.zip)

skarg, 2022-08-05:
It appears from the capture that the MAC=4 is not receiving valid replies.

In the POLL_FOR_MASTER state, the node listens for a reply to a previously sent Poll For Master frame in order to find a successor node, and then immediately sends a Token frame to the successor.  This is not happening in the capture.

Could be anything (bad MSTP device/OS timing/UARTs/transceiver, components around the transceiver).  If related to RS485 cable: it could be missing termination, wrong wire type (capacitance), bad wiring, inverted polarity, poor connections, bad ground/reference, noise (cable near strong EMI noise).

raghavan97, 2022-08-07:
Thanks a lot, Steve. We are trying to look at the Electrical aspects more deeply now. 

Having said that, what I am not able to fully get my head around is , how is our device able to "see" those responses , while the MAC=4 is not able to see it ? I mean it is the same wire. If the Wire had incorrect electrical characteristics (like wrong termination, incorrect polarity) , then none of us should be seeing the responses. What do you reckon ?

I am pretty sure, there is something that I am missing here ?

Thanks a lot for the response, Steve. We have got another direction to look at.

skarg, 2022-08-08:
A more detailed FAQ on RS485 troubleshooting:
https://store.chipkin.com/articles/rs485-faq

"see" - the RS485 transceiver must detect voltages within a certain range, and wiring problems can cause reflections and other signal anomalies that will prevent the voltage from being detected at the proper time for a particular baud rate.  The MCU UART must also be capable of detecting the 1's and 0's that are signaled from the transceiver.

raghavan97, 2022-08-26:
Looking at Wiring. It is not an issue with the Stack

closed 2022-08-26 by raghavan97