bacnet-stack/bacnet-stack #333: bacwi does not allow mac as a command line option

raghavan97, 2022-08-26:
I have a test setup where I am running the apps from bacnet-stack to do who-is, read property and write property.

I am facing the following challenges.  It could related to my ignorance on the stack.

(1) When I run ./bacwi to do who is. I am not able to discover the device that exists locally. I changed the port at which the device is exposed to say 47809, then I am able to run 

**./bacwi --mac 192.168.11.234:47809**

and get the result

I am assuming that I am not able to discover the device because both bacwi and the device are binding to 47808. Is that the case ? can we not bind to reuse the address and get the responses ? Just asking.

Now after I discover with **./bacwi --mac 192.168.11.234:47809** , I am able to read a property using 
**./bacrp --mac 192.168.11.234:47809 78901 analog-output 116 85**

But when I try to do a write property, it does not offer me the same flexibility. We cannot do
**./bacwp --mac 192.168.11.234:47809 78901 analog-output 116 85 16 -1 4 73**

because, bacwp does not take _--mac_ command line option.

My questions are 

(1) is there a way we can run ./bacwi to discover a device that is running locally ?

(2) If (1) above is not possible, would it be a good idea to add a command line option for 'mac' in ./bacwp

Thanks.

Bye,
Raghavan V

AlexanderWells-diamond, 2022-08-26:
This is due to the BACnet protocol - as you observed, by default every device will bind to the same port.

The way to fix this is to run the server, and then run the script `./bin/bvlc.sh`. This should allow the bacwi (and other) programs to work correctly.

raghavan97, 2022-08-26:
Alex,

Thanks for the response.

I missed mentioning that the server that I am running is NOT from bacnet-stack. I am trying to use the bacnet-stack clients bacwi, bacrp, bacwp to talk to a bacnet server from a Third party.

I tried running the bvlc.sh as you suggested and then ran ./bacwi , but it did not discover the device from third party bacnet server.

Do those environment variables affect both server and client ? Since I am not running bacnet-stack server , it might not be working ?

Thanks.

skarg, 2022-08-27:
You can also use 'static binding' where the MAC address (and DNET and DADR if routed) are stored locally in the address_cache file.  The apps will look in the address_cache file for the address before attempting to send a Who-Is and receiving an I-Am for address binding.  The format of the address cache is the same format as the output of bacwi application (and detailed in the address.c module).  There are also details in Q/A-17:
https://github.com/bacnet-stack/bacnet-stack/blob/master/doc/README.faq

raghavan97, 2022-08-27:
Steve,

The static binding is a boon. It solves the issue wonderfully well. My command lines are still short and crisp.

Thanks a lot for the tip.

I am closing this issue, since we do not need to add this option. We can use the feature of static binding.

Thanks.

Raghavan V

closed 2022-08-27 by raghavan97