bacnet-stack/bacnet-stack #336: Questions about events and alarms

MightyPork, 2022-09-20:
Hi, I'm now working on receiving and acknowledging events and alarms. A lot of things are still unclear to me...

Our product is something that tries to monitor devices in a Bacnet network and log events, value changes, and give users option to acknowledge alarms. Getting to test it with an actual bacnet network is somewhat difficult, so I'm writing everything in a "best guess" way and then testing if it works later.

1. I understand that to subscribe to events, my Device ID or IP ("MAC") address must be stored in a NotificationClass, this is probably done using Desigo or similar software. Or should I be able to do that through the library? (Is that a good idea?). 

    I already subscribe to COV through bacnet stack, that works but is very slow if we have hundreds of monitored objects. There should be just a few notification classes though ... if I understood how that's supposed to be set up

2. Can I set a custom device ID when using the library as a client? There is `Device_Set_Object_Instance_Number(BACNET_MAX_INSTANCE);` that I copied from the examples, i.e. `0x3FFFFF`. Is this number special? Can I just put a custom number there (that isn't used in the EDE file) and then use it as a device ID in the NotificationClass config?

3. How to receive notifications - I didn't find an example in the apps, only acking and sending... Now, I have this, loosely built following the COV example. Does it look like it might work?

```c
void Bacw_Confirmed_Event_Notification_Handler(uint8_t *service_request,
                                               uint16_t service_len,
                                               BACNET_ADDRESS *src,
                                               BACNET_CONFIRMED_SERVICE_DATA *service_data)
{
    int len;
    BACNET_NPDU_DATA npdu_data;
    BACNET_ADDRESS my_address;
    int pdu_len = 0;
    int bytes_sent = 0;

    /* encode the NPDU portion of the packet */
    datalink_get_my_address(&my_address);
    npdu_encode_npdu_data(&npdu_data, false, MESSAGE_PRIORITY_NORMAL);
    pdu_len = npdu_encode_pdu(
            &Handler_Transmit_Buffer[0], src, &my_address, &npdu_data);

    bacw_info("cevent notify!");

    if (service_data->segmented_message) {
        len = abort_encode_apdu(&Handler_Transmit_Buffer[pdu_len],
                                service_data->invoke_id, ABORT_REASON_SEGMENTATION_NOT_SUPPORTED,
                                true);
        bacw_error("CEVENT: Segmented message. Sending Abort!");
        goto CEVENT_ABORT;
    }

    BACNET_EVENT_NOTIFICATION_DATA data = {};
    len = event_notify_decode_service_request(service_request, service_len, &data);
    if (len > 0) {
        bacw_rs_handle_event(&data);

        //
        len = encode_simple_ack(&Handler_Transmit_Buffer[pdu_len],
                                service_data->invoke_id, SERVICE_CONFIRMED_EVENT_NOTIFICATION);
        bacw_debug("CEVENT: Sending Simple Ack!");
    } else {
        bacw_debug("CEVENT: Fail to decode, sending abort");
        len = abort_encode_apdu(&Handler_Transmit_Buffer[pdu_len],
                                service_data->invoke_id, ABORT_REASON_OTHER, true);
        goto CEVENT_ABORT;
    }

CEVENT_ABORT:;
    pdu_len += len;

    bytes_sent = datalink_send_pdu(
            src, &npdu_data, &Handler_Transmit_Buffer[0], pdu_len);

    if (bytes_sent <= 0) {
        bacw_error("CEVENT: Failed to send PDU (%s)!", strerror(errno));
    }
}

void Bacw_Unconfirmed_Event_Notification_Handler(
        uint8_t *service_request, uint16_t service_len, BACNET_ADDRESS *src)
{
    int len;
    bacw_info("uevent notify!");

    BACNET_EVENT_NOTIFICATION_DATA data = {};
    len = event_notify_decode_service_request(service_request, service_len, &data);
    if (len > 0) {
        bacw_rs_handle_event(&data);
    }
}
```

and to bind the handlers:

```c
    /* Events */
    apdu_set_confirmed_handler(
        SERVICE_CONFIRMED_EVENT_NOTIFICATION, Bacw_Confirmed_Event_Notification_Handler);
    apdu_set_unconfirmed_handler(
        SERVICE_UNCONFIRMED_EVENT_NOTIFICATION, Bacw_Unconfirmed_Event_Notification_Handler);
```

skarg, 2022-10-27:
For BACnet client (initiation) clarity, I usually turn to ASHRAE/ANSI 135.1 Method of Test for Conformance to BACnet.
```
8. APPLICATION SERVICE INITIATION TESTS ....................................................................................................... 190
8.1 AcknowledgeAlarm Service Initiation Tests ........................................................................................................ 190
8.2 ConfirmedCOVNotification Service Initiation Tests ........................................................................................... 191
8.3 UnconfirmedCOVNotification Service Initiation Tests ....................................................................................... 200
8.4 ConfirmedEventNotification Service Initiation Tests .......................................................................................... 203
8.5 UnconfirmedEventNotification Service Initiation Tests ...................................................................................... 245
8.6 GetAlarmSummary Service Initiation Tests ......................................................................................................... 265
8.7 GetEnrollmentSummary Service Initiation Tests ................................................................................................. 266
8.8 GetEventInformation Service Initiation Tests ...................................................................................................... 267
```
There is also guidance in the ASHRAE/ANSI 135 BACnet standard, and also from the BTL Implementation Guidelines.  See:
https://btl.org/testing-resources/

MightyPork, 2022-11-30:
We purchased "ANSI/ASHRAE Standard 135-2020", so some things are more clear to me now. In particular, I think I will need to implement AddListElement and RemoveListElement to manage the RecipientList in a NotificationClass. The list can be read using regular ReadProperty or ReadRange - not sure if actual decoding and encoding is implemented? The readprop app prints some fields from the payload.

I'm out of luck with finding the implementation in jbennet this time, right...?

closed 2023-10-11 by MightyPork