bacnet-stack/bacnet-stack #341: Different optional properties

Wielebny666, 2022-09-28:
Whether the current bacnet-stack allows the creation of multiple objects of the same type but with different optional properties, ie an instance_object dependent property list.

regards

skarg, 2022-09-29:
The current API for the property lists expects all instances to support the same properties.
```
typedef void (
    *rpm_property_lists_function) (
    const int **pRequired,
    const int **pOptional,
    const int **pProprietary);
```

I've made products with this library, but changed the API for the property lists like this:
```
/** Fetches the lists of properties (array of BACNET_PROPERTY_ID's) for this
 *  object type, grouped by Required, Optional, and Proprietary.
 * A function template; @see device.c for assignment to object types.
 * @ingroup ObjHelpers
 *
 * @param object_instance [in] which object instance is being queried
 * @param pRequired [out] Pointer reference for the list of Required properties.
 * @param pOptional [out] Pointer reference for the list of Optional properties.
 * @param pProprietary [out] Pointer reference for the list of Proprietary
 *                           properties for this BACNET_OBJECT_TYPE.
 */
typedef void (*rpm_property_lists_function) (
    uint32_t object_instance,
    const int **pRequired,
    const int **pOptional,
    const int **pProprietary);
```
We should plan to make this API change in the next major revision.

Wielebny666, 2022-10-04:
Thank you for your response,
when is this version planned?