bacnet-stack/bacnet-stack #346: All-zero BACnet IP Listening Problem

Lee-nju, 2022-10-09:
This is what the source code in version bacnet-stack-v1.0.0 ports/linux/bip-init.c says, which binds ip 0.0.0.0.
[image]

After the protocol is opened，BACnet/IP is listening ip all-zero which creates an implicit risk.
```sh
$ netstate -anp
```
[image]

sambabug, 2022-10-09:
Is it because the socket is already bound to another address (EINVAL)? Is this reproducible on other machines?

skarg, 2022-10-12:
The purpose of using INADDR_ANY is to bind to all network interfaces with the specific BACnet/IP UDP port number.

See:
https://stackoverflow.com/questions/16508685/understanding-inaddr-any-for-socket-programming

> If the (sin_addr.s_addr) field is set to the constant INADDR_ANY, as defined in netinet/in.h, the caller is requesting that the socket be bound to all network interfaces on the host. Subsequently, UDP packets and TCP connections from all interfaces (which match the bound name) are routed to the application. This becomes important when a server offers a service to multiple networks. By leaving the address unspecified, the server can accept all UDP packets and TCP connection requests made for its port, regardless of the network interface on which the requests arrived.

The master branch code (soon to be 1.1.0) uses two sockets: one for unicast, and one for broadcast.

Lee-nju, 2022-10-14:
Well, thank you for your patience, and I understand the reason for this. But when it comes to security, that's not what all users want. It is expected that unicast will be integrated in the next version.

Lee-nju, 2022-11-03:
Let me ask you a question. For example, I have a device with multiple network ports, some of which do not want to run bacnet Ip. What can I do? Is it possible to open bip-init.c, like dlmstp_linux.c, so that developers can define their own networks?

skarg, 2022-11-04:
The environment variable BACNET_IFACE is used to explicitly tell bip-init.c which interface to use without having to modify the code.
$ export BACNET_IFACE=eth0

Lee-nju, 2022-11-09:
@skarg I did bind GE0 port to IFACE, but the final result was as shown in the problem description.
[image]
I found that it was modified in this code.

PR #489 (2023-09-10): fix BACnet/IP on OS to bind broadcast to specific port
Fixes #346 by binding the broadcast socket to the port specific broadcast address rather than INADDR_ANY or INADDR_BROADCAST.  Added additional compile options for those who might need them.  For Linux/BSD:
```
#ifdef BACNET_IP_BROADCAST_USE_INADDR_ANY
    sin.sin_addr.s_addr = htonl(INADDR_ANY);
#elif BACNET_IP_BROADCAST_USE_INADDR_BROADCAST
    sin.sin_addr.s_addr = htonl(INADDR_BROADCAST);
#else
    sin.sin_addr.s_addr = BIP_Broadcast_Addr.s_addr;
#endif
```
For Windows:
```
#else
    sin.sin_addr.s_addr = BIP_Address.s_addr;
#endif
``` [... cut]
commit 55b8b3d9 (2023-09-10): fix BACnet/IP on OS to bind broadcast to specific port (#489)

* Fixes #346 by binding the broadcast socket to the port specific broadcast address rather than INADDR_ANY or INADDR_BROADCAST. Added additional compile and environment options for those who might need them. 

* Changed the define USE_CLASSADDR to BACNET_IP_BROADCAST_USE_CLASSADDR to help manage namespaces.

* Added BACNET_IP_BROADCAST_BIND_ADDR environment variable to override the default broadcast binding address.

---------
closed 2023-09-10 by skarg