bacnet-stack/bacnet-stack #365: Help on running bacwi and bacepics on the same machine

raghavan97, 2022-12-06:
Hi All,

I have got an BACnet server application running on our device A. When I run `netstat -a | grep 47808` , I see that it is bound to all the interfaces.

```
admin@JKTGG92:~$ netstat -a | grep 47808
udp        0      0 *:47808                 *:*
udp        0      0 192.168.11.255:47808    *:*
udp        0      0 JKTGG92:47808           *:*

```

I have another device B on the same physical network. I am able to run 'bacwi' and 'bacepics' . It shows all the information from device A in a very nice way.

```
admin@Riptide-53-Test-BE:~$ ./bacwi
;Device   MAC (hex)            SNET  SADR (hex)           APDU
;-------- -------------------- ----- -------------------- ----
  78901   C0:A8:0B:EA:BA:C0    0     00                   1476
  112100  C0:A8:0B:EA:BA:C0    1     01                   1024
  112409  C0:A8:0B:EA:BA:C0    1     02                   1024
;
; Total Devices: 3

```
```
admin@Riptide-53-Test-BE:~$ ./bacepics -v 112409
PICS 0
BACnet Protocol Implementation Conformance Statement

--
--
-- Generated by BACnet Protocol Stack library EPICS tool
-- BACnet/IP Interface for BACnet-stack Devices
-- http://sourceforge.net/projects/bacnet/
--
--

Vendor Name: "Riptide IO"
Product Name: "Brightedge Virtual Device"
Product Model Number: "Brightedge Virtual Device"
Product Description: "your product description here"

```
I dont have the luxury of having 2 devices in our deployment. I would like to run both bacwi and bacepics on the same device.

I tried to run it and it comes back with APDU timeout. I tried a few options 

Loopback
`./bacwi -t 7F:00:00:01:BA:C0`

directed mac
`./bacwi --mac 192.168.11.234:47808`

specific dnet
`./bacwi --mac 192.168.11.234 --dnet 1`

With bacepics, I tried changing the sport

changing sport
`./bacepics -p 6000 -t C0:A8:0B:EA:BA:C0 112409
`

All of the above yielded nothing.

Am I missing something ?

Is it possible to run these commands on the same box and get the results ?

Thanks for the help.

AlexanderWells-diamond, 2022-12-06:
You have to set some special environment variables to run two Bacnet devices on the same machine.

Try starting the server, then running [this script](https://github.com/bacnet-stack/bacnet-stack/blob/master/bin/bvlc.sh), then running the `bacwi` command. There's some docs for that script in the README in the same folder.

raghavan97, 2022-12-06:
Thanks. I did this

`export BACNET_IFACE=eth1`

and then ran the `./bacwi `, it worked great
I tried `./bacepics` too. It worked fine

Thanks a lot, Alex.