bacnet-stack/bacnet-stack #370: ESP32 BACnet through Wifi

maialbe2823, 2022-12-23:
Hi, I am currentling trying to make ESP32 a BACnet device. However; after more than 2 weeks I cann seem to make it stable. I am able to see the ESP32 in a bacnet network for a few seconds and then it goes offline. The ESP32 just keep saying: handler_who is and then is goes offline. I do see that my BACnet explorer get the ESP32 device number but just that. It doesnt get to read the ESP32 device name or any service or objects.
Please could someone guide into what I am missing in order to stablish a stable connection between my BACnet explorer and my ESP32 device plus getting all the ESP32 objects and services. Thanks

MoreTore, 2023-02-11:
ESP32 is not a good platform. The debugger is not stable so it is difficult to debug.

drwho100, 2023-02-13:
> ESP32 is not a good platform. The debugger is not stable so it is difficult to debug.

I used serial.print() for debug. Don't see any problems with this chip. It worked very fast performance compare to others chipset.

MoreTore, 2023-02-13:
Yea sure you can do printed but if you want to single step the code it becomes unstable. Makes it difficult to figure out tricky bugs.

drwho100, 2023-02-15:
> Yea sure you can do printf but if you want to single step the code it becomes unstable. Makes it difficult to figure out tricky bugs.

just print variables of the part unstable out including loop or array with space or slash.

lukedukeus, 2023-02-24:
@maialbe2823 Did you make any progress? I am trying to do the same thing, but running into other issues.