bacnet-stack/bacnet-stack #398: bacnet-tools v1.0 bacserv | Tag value out of range

tblong, 2023-02-15:
Using bacnet-tools v1.0 `bacserv` playing the role of the BACnet device for this question.  I'm currently dealing in the BACnet protocol revision 19 and I've not dug into whether or not unsigned int tag lengths greater than 4 are valid in this and later revisions.

**The question:** Is it by design to have `bacserv` respond with invalid application tag lengths (5+) for the unsigned int type on Accumulator instances 32 and greater? 

If the above is intentional, then I have to imagine this would for the purpose of testing the resiliency of other devices and tools?  If not, then below is a capture showing the application tag length of 5 being returned from reading ACC:33 served up by the `bacserv` application:

[image]

skarg, 2023-02-16:
It is by design to have the example/sample/demo/basic Accumulator object have a present-value that is 1-bit to 64-bits to see how clients respond to unsigned values larger than 32-bits.  The BACnet standard allows integer values larger than 32-bits, although some of the object property and service values are constrained.

tblong, 2023-02-17:
Thanks much Steve, I'll close this question out.

closed 2023-02-17 by tblong