bacnet-stack/bacnet-stack #409: ESP32 MSTP Port Issue No data for us

maialbe2823, 2023-03-12:
I’m trying to make esp32 port bacnet mstp but it giving a hard time. I have working for both Ethernet and wifi. But after 2 weeks of trying to add mstp to it all I get is the other controllers data and I not able to send data. Plus after a while the CRC goes bad. Any suggestions on how to address this? Or any is there any working version of a dlmstp and rs485 for esp32?
Thanks

TadhgHwang, 2023-03-13:
Have you seen "https://github.com/KaiDroste/bacnet-stack-esp32.git" under ports/esp-mstp, I tested this example and found something in Yabe, still trying to understand what happened in dlmstp.c

maialbe2823, 2023-03-13:
Yes I have, but not luck on it. It just keeps searching for available data.