bacnet-stack/bacnet-stack #410: reduce footprint of device.c file

mczerski, 2023-03-13:
Hi,
because file src/bacnet/basic/object/device.c defines My_Object_Table array with all possible object types and add some of them in Device_Init to use this file in my project I need to add implementation files of all used objects even if I do not use My_Object_Table (passing something as object_table argument for Device_Init). Because I would like to run bacnet stack on stmf103c8 I cannot add all of the implementation due to program memory restrictions.
I see two possible solutions, do not put all object types into My_Object_Table (just the device object which is the bare minimum required by the standard) or define the My_Object_Table somewhere else.

skarg, 2023-03-13:
I haven't figured out a nice way to have generic device object that is referenced by the required and optional (example) service handlers.  I would expect that any product would have their own device.c (or whatever the build wants to call it) that includes children objects and service handler function calls.

See other example device.c and their objects used in example microcontroller builds:
https://github.com/bacnet-stack/bacnet-stack/blob/master/ports/stm32f10x/device.c
https://github.com/bacnet-stack/bacnet-stack/blob/master/ports/stm32f4xx/device.c
https://github.com/bacnet-stack/bacnet-stack/blob/master/ports/at91sam7s/device.c
https://github.com/bacnet-stack/bacnet-stack/blob/master/ports/xplained/device.c
etc

mczerski, 2023-03-14:
yah, I see that in every project in port folder copies those object implementation files. seems like a lot of work to implement so simple stuff like binary input or output. and then a lot of work to follow the stack developement.
What exactly is not generic about device.c ?? I just modified port/stmf10x to use all object implementations from src/bacnet/basic/object and it seems like everything is working.