bacnet-stack/bacnet-stack #411: reduce footprint of BACNET_APPLICATION_DATA_VALUE struct

mczerski, 2023-03-14:
Hi,

in src/bacnet/bacapp.h in BACNET_APPLICATION_DATA_VALUE struct there is a union called type. The types in that union are possible to enable / disable through defines. Most of the types have separate define for them but there is one define BACAPP_TYPES_EXTRA which enables a lot of types and especially BACNET_WEEKLY_SCHEDULE which is HUGE in embedded environment. It would be nice if this BACNET_WEEKLY_SCHEDULE type had separate define to enable/disable it.

skarg, 2023-03-14:
The BACAPP_TYPES_EXTRA is for application datatypes that are not primitive datatypes enumerated by BACnet135-xxxx-20.2 Encoding the Variable Part of BACnet APDUs.  I had an implementation with EVERY datatype ifdef, but that got unwieldy, so I figured that BACAPP_TYPES_EXTRA could suffice.  Maybe not.

Recently I was watching [Uncle Bob talk about some Clean Code](https://youtu.be/7EmboKQH8lM) techniques, and he mentioned one of his quests: eliminate switch statements since they collect into modules that require constant modification when extensions are needed. Perhaps there is a better way to add each datatype encode/decode/parse/value that a project needs using a 'bacapp_datatype_add()' function that is a linked list of each of those members (functions, data store) that an encoder/decoder/parser needs for a specified or complex datatypes.

mczerski, 2023-03-15:
Well, quick solutions for my problem would be to just add one more define just for weekly schedule which would be consistent with current implementation.
But if You thinking about larger refactor I would suggest to use object oriented paradgim. With c++ it would be much easier, but with c it is also possibile (gobject library is one example of doing object oriented programming in c). That way You could implement different data types as specialisations of the abstract interface with logic for encode/decode/parse/value and keep those specialisations in separate modules.

closed 2023-12-01 by skarg