bacnet-stack/bacnet-stack #412: how to close bacnet ip socket ?

cyberumut, 2023-03-15:
I have a master device that constantly accesses a device and reads its analog output 3 address, but I get an error for a certain time (No buffer space available). I used readprop demo codes

how can i fix tih problem ?

skarg, 2023-03-15:
When apps/readprop uses the BACnet/IP sockets, the bip-init.c opens the socket(s), performs some sending and receiving, and then closes the socket with the atexit() function which calls bip_cleanup().

Have you adapted the readprop application?

cyberumut, 2023-03-15:
> When apps/readprop uses the BACnet/IP sockets, the bip-init.c opens the socket(s), performs some sending and receiving, and then closes the socket with the atexit() function which calls bip_cleanup().
> 
> Have you adapted the readprop application?

yes i adapted since 2019.

i wanna readprop device id 1938 and analog output 3 . 

i can running func readpro(1938,3)

`
int readprop(int addr,int point)
{

    static uint8_t Rx_Buf[MAX_MPDU] = { 0 };

    BACNET_ADDRESS src = {
        0
    };  /* address where message came from */
    uint16_t pdu_len = 0;
    unsigned timeout = 100;     /* milliseconds */
    unsigned max_apdu = 0;
    time_t elapsed_seconds = 0;
    time_t last_seconds = 0;
    time_t current_seconds = 0;
    time_t timeout_seconds = 0;
    bool found = false;
    long dnet = -1;
    BACNET_MAC_ADDRESS mac = { 0 };
    BACNET_MAC_ADDRESS adr = { 0 };
    BACNET_ADDRESS dest = { 0 };
    bool specific_address = false;

                Target_Device_Object_Instance=addr;
   
                Target_Object_Type = 1;

                Target_Object_Instance = point;

                Target_Object_Property = 85;
    
		
        
    if (Target_Device_Object_Instance > BACNET_MAX_INSTANCE) {
        fprintf(stderr, "device-instance=%u - it must be less than %u\n",
            Target_Device_Object_Instance, BACNET_MAX_INSTANCE);
        
    }
    address_init();
    /* setup my info */
    Device_Set_Object_Instance_Number(BACNET_MAX_INSTANCE);
    Init_Service_Handlers();
    dlenv_init();
    atexit(datalink_cleanup);
    atexit(bip_cleanup);
    /* configure the timeout values */
    last_seconds = time(NULL);
    timeout_seconds = (apdu_timeout() / 1000) * apdu_retries();
    /* try to bind with the device */
    found = address_bind_request(Target_Device_Object_Instance, &max_apdu,&Target_Address);
    if (!found) {
        Send_WhoIs(Target_Device_Object_Instance,
            Target_Device_Object_Instance);
    }
    /* loop forever */
    for (;;) {

        //printf("for ici \n");
        /* increment timer - exit if timed out */
        current_seconds = time(NULL);

        /* at least one second has passed */
        if (current_seconds != last_seconds)
            tsm_timer_milliseconds((uint16_t) ((current_seconds -
                        last_seconds) * 1000));
        if (Error_Detected)
            break;
        /* wait until the device is bound, or timeout and quit */
        if (!found) {
            found =
                address_bind_request(Target_Device_Object_Instance, &max_apdu,
                &Target_Address);
        }
        if (found) {
            if (Request_Invoke_ID == 0) {
                        //printf("invode id  ici \n");
                Request_Invoke_ID =
                    Send_Read_Property_Request(Target_Device_Object_Instance,
                    Target_Object_Type, Target_Object_Instance,
                    Target_Object_Property, Target_Object_Index);
            } else if (tsm_invoke_id_free(Request_Invoke_ID))
                break;
            else if (tsm_invoke_id_failed(Request_Invoke_ID)) {
                fprintf(stderr, "\rError: TSM Timeout!\n");
                tsm_free_invoke_id(Request_Invoke_ID);
                Error_Detected = true;
                /* try again or abort? */
                break;
            }
        } else {
            /* increment timer - exit if timed out */
            elapsed_seconds += (current_seconds - last_seconds);
            if (elapsed_seconds > timeout_seconds) {
                printf("\rError: APDU Timeout!\n");
                Error_Detected = true;
                break;
            }
        }

        /* returns 0 bytes on timeout */
        pdu_len = datalink_receive(&src, &Rx_Buf[0], MAX_MPDU, timeout);
		
        /* process */
      /**/  if (pdu_len) {
            npdu_handler(&src, &Rx_Buf[0], pdu_len);
        }

		

        /* keep track of time for next check */
        last_seconds = current_seconds;
    }
Request_Invoke_ID= 0;
bip_cleanup();
    return 0;
}`

cyberumut, 2023-03-15:
> When apps/readprop uses the BACnet/IP sockets, the bip-init.c opens the socket(s), performs some sending and receiving, and then closes the socket with the atexit() function which calls bip_cleanup().
> 
> Have you adapted the readprop application?
 error 
Failed to Send Who-Is Request (Too many open files in system)!

skarg, 2023-03-15:
I'm not seeing your entire application, so I can only guess: your application should only be calling some functions (typically the init() functions) once, those functions that are called above the forever loop.  Calling dlenv_init() - which calls datalink_init() -  without directly calling datalink_cleanup() at the end of the function - during the application will create new sockets without removing the sockets.

For an example of ReadProperty from within a server application, you could be using a newly introduced module for R/W:
https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/client/bac-rw.c

You can see how it works from the example application:
https://github.com/bacnet-stack/bacnet-stack/blob/master/apps/server-client/main.c
which also makes use of:
https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/client/bac-task.c
https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/client/bac-data.c

cyberumut, 2023-03-15:
> I'm not seeing your entire application, so I can only guess: your application should only be calling some functions (typically the init() functions) once, those functions that are called above the [...]
thank u problem solved 
i did move init() to handlers in and just running 1  in app_main