bacnet-stack/bacnet-stack #415: I'm unable to receive and APDU data when using MSTP

kaleidoscope-dev, 2023-03-20:
Hello,

I've compiled the bacnet-stack library with MSTP option for Windows10 in VS2019 and I can't see any devices that respond with I-AM.
I've used other programs (such as YABE) and I can confirm that devices are there and communicating, but not when using who-is example in the library. 
I've also added some printf in the MasterNodeStateMachine and I can't see anything coming or going. (not even tokens?). 
After some time the Master goes into NO_TOKEN state and start issuing PFM, but very slow (between 500-1000ms) and never until the end (127).

Here is the output with some print state I get.

PS C:\dev\prj\Windows\BACnet\KargStack\bacnet-stack-1.0.0\Debug> .\whois.exe
max_masters  = 127
RS485 Interface: COM3
MSTP_Init: This_Station: 1
MS/TP MAC: 01
MS/TP Max_Master: 7F (127)
MS/TP Max_Info_Frames: 1
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: USE_TOKEN
MASTER: IDLE
NO TOKEN: silence_timer: 510
Sending Next PFM Poll (3)
Sending Next PFM Poll (4)
Sending Next PFM Poll (5)
Sending Next PFM Poll (6)
Sending Next PFM Poll (7)
Sending Next PFM Poll (8)
Sending Next PFM Poll (9)
Sending Next PFM Poll (10)
Sending Next PFM Poll (11)
Sending Next PFM Poll (12)
Sending Next PFM Poll (13)
Sending Next PFM Poll (14)
Sending Next PFM Poll (15)
Sending Next PFM Poll (16)
Sending Next PFM Poll (17)
Sending Next PFM Poll (18)
Sending Next PFM Poll (19)
Sending Next PFM Poll (20)
Sending Next PFM Poll (21)
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: USE_TOKEN
MASTER: IDLE
NO TOKEN: silence_timer: 510
Sending Next PFM Poll (3)
Sending Next PFM Poll (4)
Sending Next PFM Poll (5)
Sending Next PFM Poll (6)
Sending Next PFM Poll (7)
Sending Next PFM Poll (8)
Sending Next PFM Poll (9)
Sending Next PFM Poll (10)
Sending Next PFM Poll (11)
Sending Next PFM Poll (12)
Sending Next PFM Poll (13)
Sending Next PFM Poll (14)
Sending Next PFM Poll (15)
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: IDLE
MASTER: USE_TOKEN
MASTER: IDLE
NO TOKEN: silence_timer: 510
Sending Next PFM Poll (3)
Sending Next PFM Poll (4)
Sending Next PFM Poll (5)
Sending Next PFM Poll (6)
Sending Next PFM Poll (7)
Sending Next PFM Poll (8)
Sending Next PFM Poll (9)
Sending Next PFM Poll (10)
Sending Next PFM Poll (11)
Sending Next PFM Poll (12)
Sending Next PFM Poll (13)
Sending Next PFM Poll (14)
;Device   MAC (hex)            SNET  SADR (hex)           APDU
;-------- -------------------- ----- -------------------- ----
;
; Total Devices: 0

I also Have a question about TOKEN passing. When masters encounter one another, shouldn't they just start exchanging tokens?  Like 
GOT_TOKEN: Nothing to send -> DONE_WITH_TOKEN -> PASS_TOKEN -> GOT_TOKEN etc?

Protocol Standard kind of say something like that If I understood it correctly.   
Also, Protocol states that ReceiveFSM an  MasterFSM and Send use timers but it's unclear it they are the same timer or specific for each State Machine. Stack as far as I can see uses the same for ReceiveFSM and MasterFSM but different for Send?

Thanks,
Dario

skarg, 2023-09-09:
I would expect to see PFM (poll-for-master) followed by a RPFM (reply-to-poll-for-master) followed by a Token pass from Manager/Master to the peer device (NS - next station).  Master/Manager nodes discover each other using PFM/RPFM before they start exchanging tokens, and the node only really cares about the NS (next-station) with the next highest MAC address (in the ring, so it wraps around).  The Master/Manager will periodically check with PFM until until the NS is the next highest MAC address.

You should be able to use Wireshark to monitor the RS485 and MSTP to find the messages and timings.  Usually on Windows the MSTP timings are exceeded.

kaleidoscope-dev, 2024-05-05:
Hi Steve,

You are correct. This is a Windows specific timing issue. One of the sates fails to trigger and MSM resets. Linux implementations works normally.