bacnet-stack/bacnet-stack #417: BACnet Abort: Segmentation Not Supported

cyberumut, 2023-04-02:
im using bacrpm.exe 31 8 31 8 command but segmentation not supported  returned 

how can i get objects with bacrpm or different method ??

skarg, 2023-04-02:
The target device indicates that there is too much data to be returned in a single packet, and doesn't support segmentation.  To work around devices that don't support segmentation, ReadProperty or ReadPropertyMultiple needs each required property as specified by the BACnet standard (the indicated property 8 is all the required properties). 

The BACnet EPICs tool (bacepics) uses a fallback of ReadProperty for failed ReadPropertyMultiple requests for properties, and uses a [property.c](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/property.c) that enumerates all the required and optional properties for most object types.

Sometimes BACnet Arrays and Lists will exceed the size that can be returned by a target device, and those can be read one element at a time, with element zero returning the count, and subsequent elements returning the value.  BACnet Lists would need to use ReadRange service.

cyberumut, 2023-04-02:
> Collaborator

thank u Steve

closed 2023-04-03 by cyberumut