bacnet-stack/bacnet-stack #469: Which states of the master have tokens？

Twillpower, 2023-08-11:
typedef enum {
    MSTP_MASTER_STATE_INITIALIZE = 0,
    MSTP_MASTER_STATE_IDLE = 1,
    MSTP_MASTER_STATE_USE_TOKEN = 2,
    MSTP_MASTER_STATE_WAIT_FOR_REPLY = 3,
    MSTP_MASTER_STATE_DONE_WITH_TOKEN = 4,
    MSTP_MASTER_STATE_PASS_TOKEN = 5,
    MSTP_MASTER_STATE_NO_TOKEN = 6,
    MSTP_MASTER_STATE_POLL_FOR_MASTER = 7,
    MSTP_MASTER_STATE_ANSWER_DATA_REQUEST = 8
} MSTP_MASTER_STATE;

skarg, 2023-08-12:
States in which this station has the token:
- MSTP_MASTER_STATE_INITIALIZE - no
- MSTP_MASTER_STATE_IDLE - no
- MSTP_MASTER_STATE_USE_TOKEN - yes
- MSTP_MASTER_STATE_WAIT_FOR_REPLY - yes
- MSTP_MASTER_STATE_DONE_WITH_TOKEN - yes
- MSTP_MASTER_STATE_PASS_TOKEN - yes
- MSTP_MASTER_STATE_NO_TOKEN - no
- MSTP_MASTER_STATE_POLL_FOR_MASTER - yes
- MSTP_MASTER_STATE_ANSWER_DATA_REQUEST - no

Twillpower, 2023-08-12:
thank you, sir