bacnet-stack/bacnet-stack #473: Is it necessary that the slave cannot respond to WhoIs? I made some modifications to make it responsive, but something went wrong.

Twillpower, 2023-08-19:
I'm currently testing with five esp-32 devices running bacnet ms/tp, one as master and four as slaves. Since the original MSTP_Slave_Node_FSM() cannot respond to WhoIs, I have to know the device_id of the slave in advance every time, which makes me feel very inflexible, so I slightly modify it (here I refer to the stm32f10x routine):
static void MSTP_Slave_Node_FSM(void)
{
    /* destination address */
    uint8_t destination;
    /* source address */
    uint8_t source;
    /* any data to be sent - may be null */
    uint8_t * data;
    /* amount of data to be sent - may be 0 */
    uint16_t data_len;
    /* packet from the PDU Queue */
    struct mstp_pdu_packet * pkt;

    if (MSTP_Flag.ReceivedInvalidFrame)
    {
        /* ReceivedInvalidFrame */
        /* invalid frame was received */
        MSTP_Flag.ReceivedInvalidFrame = false;
    }
    else if (MSTP_Flag.ReceivedValidFrame)
    {
        ESP_LOGW("Rec FrameType", "%d", FrameType);
        switch (FrameType)
        {
            case FRAME_TYPE_BACNET_DATA_EXPECTING_REPLY:
                /* indicate successful reception to higher layer */
                if (DestinationAddress != MSTP_BROADCAST_ADDRESS)
                {
                    MSTP_Flag.ReceivePacketPending = true;
                    /* The ANSWER_DATA_REQUEST state is entered when a  */
                    /* BACnet Data Expecting Reply, a Test_Request, or  */
                    /* a proprietary frame that expects a reply is received. */
                    pkt = (struct mstp_pdu_packet *)Ringbuf_Peek(&PDU_Queue);
                    if (pkt != NULL)
                    {
                        MSTP_Send_Frame(FRAME_TYPE_BACNET_DATA_EXPECTING_REPLY, pkt->destination_mac, This_Station,
                                        (uint8_t *)&pkt->buffer[0], pkt->length);
                        /* clear our flag we were holding for comparison */
                        MSTP_Flag.ReceivedValidFrame = false;
                        /* clear the queue */
                        (void)Ringbuf_Pop(&PDU_Queue, NULL);
                    }
                    else if (rs485_silence_elapsed(Treply_delay))
                    {
                        /* If no reply will be available from the higher layers
                           within Treply_delay after the reception of the final
                           octet of the requesting frame (the mechanism used
                           to determine this is a local matter), then no reply
                           is possible. */
                        MSTP_Flag.ReceivedValidFrame = false;
                    }
                }
                else
                {
                    /* no reply when addressed as Broadcast */
                    MSTP_Flag.ReceivedValidFrame = false;
                }
                break;
            case FRAME_TYPE_TEST_REQUEST:
                MSTP_Flag.ReceivedValidFrame = false;
                destination                  = SourceAddress;
                source                       = This_Station;
                data                         = &InputBuffer[0];
                data_len                     = DataLength;
                MSTP_Send_Frame(FRAME_TYPE_TEST_RESPONSE, destination, source,
                                data, data_len);
                break;
            case FRAME_TYPE_BACNET_DATA_NOT_EXPECTING_REPLY:
                if ((DestinationAddress == MSTP_BROADCAST_ADDRESS) &&
                    (npdu_confirmed_service(InputBuffer, DataLength)))
                {
                    /* BTL test: verifies that the IUT will quietly
                       discard any Confirmed-Request-PDU, whose
                       destination address is a multicast or
                       broadcast address, received from the
                       network layer. */
                    ESP_LOGD("TAG", "Testing");
                    MSTP_Flag.ReceivedValidFrame = false;
                }
                else if ((DestinationAddress == This_Station) || (DestinationAddress == MSTP_BROADCAST_ADDRESS))
                {
                    if (!Ringbuf_Empty(&PDU_Queue))
                    {
                        pkt = (struct mstp_pdu_packet *)Ringbuf_Peek(&PDU_Queue);
                        MSTP_Send_Frame(FRAME_TYPE_BACNET_DATA_NOT_EXPECTING_REPLY, pkt->destination_mac, This_Station,
                                        (uint8_t *)&pkt->buffer[0], pkt->length);
                        (void)Ringbuf_Pop(&PDU_Queue, NULL);
                    }
                    /* indicate successful reception to higher layer */
                    MSTP_Flag.ReceivePacketPending = true;
                    MSTP_Flag.ReceivedValidFrame   = false;
                }
                break;
            default:
                MSTP_Flag.ReceivedValidFrame = false;
                break;
        }
        MSTP_Flag.ReceivedValidFrame = false;
    }
}
After the modification, the slave can reply WhoIs, and the master can also read the attributes of the slave, but the master does not receive I_am and attribute values every time, and in most cases it cannot receive it. I try to check the log of the slave device , I found that their sending is normal, is it caused by my incorrect handling?

skarg, 2023-08-19:
Yes, it is required that MSTP Slave / Subordinate devices shall not respond to Who-Is requests.  It is better that devices, which are capable of being MSTP Master / Manager devices, should not be MSTP Slave / Subordinate devices so that they can participate in Token passing and be able to respond to broadcast requests.

The BACnet application layer (where Who-Is/I-Am originate) does not care about the datalink (much), and any device can be a client/server or server, which is not related to MSTP Master/Slave in the datalink layer.

Note: There is Who-Is proxy service that was added to BACnet in protocol-revision 4 that standardized MSTP Slave discovery (typically within a Router). See page 15/16 of Addendum 135-2001a:
https://bacnet.org/wp-content/uploads/sites/4/2022/08/Add-2001-135a.pdf

Twillpower, 2023-08-20:
I don’t know if my understanding is wrong. Only the master device can join the token ring, and only when one of the master devices gets the token can it actively send requests to other master devices, and the slave device will not join the token ring. Card ring, it can only receive requests and respond.
If so, then when all devices are masters, one of the devices must wait for the token to read the attributes of other devices. Based on this, I want to implement a non-peer-to-peer network with one master and multiple slaves , so that the master device can read the properties of other devices at any time without waiting for the token

skarg, 2023-08-20:
As I mentioned, it is easy to confuse MSTP master/slave names with BACnet client/server.

BACnet ReadProperty services use client/server application layer that sits above the network and datalink layer.  All BACnet devices are servers (must support ReadProperty service and a Device object) and some devices are also client (can use ReadProperty service to request another device properties).

The BACnet client/server application layer uses whatever the network datalink - BACnet MSTP or BACnet/IP or BACnet/IPv6 or BACnet/SC or ARCNET etc - and routers can route the exact same APDU (i.e. ReadProperty service) between any of the datalinks.

BACnet MSTP datalink uses a token passing mechanism to manage single access on RS485 shared bus, similar to ARCNET156 on RS485. If a BACnet MSTP node is unable to participate in passing the token, it is referred to as a slave/secondary node, but this also breaks the ability for that node to be a BACnet client (it never gets the Token) or be able to respond to anything other than a Data Expecting Reply frame (per specification).