bacnet-stack/bacnet-stack #475: When I call (handler_cov_task();) in bacnetTask, stack overflow will always occur.

Twillpower, 2023-08-23:
I adjusted the stack size to 10000 or even larger, with a priority of 20, and stack overflow still occurs.
void bacnet_task(void)
{
    uint16_t pdu_len;
    BACNET_ADDRESS src; /* source address */

    /* handle the communication timer */
    if (mstimer_expired(&DCC_Timer))
    {
        mstimer_reset(&DCC_Timer);
        dcc_timer_seconds(DCC_CYCLE_SECONDS);
    }
    handler_cov_timer_seconds(1);
    // test
    HBD_LAMP_Is_People_Set(0, wy_is_people);
    count++;
    if(count >= 500)
    {
        wy_is_people = !wy_is_people;
    }
// test
    /* handle the messaging */
    pdu_len = datalink_receive(&src, &PDUBuffer[0], MAX_MPDU, 0);
    if (pdu_len)
    {
        // esp_log_buffer_hex("datalink_receive", &PDUBuffer[0], pdu_len);
        npdu_handler(&src, &PDUBuffer[0], pdu_len);
    }
    handler_cov_task();
}

Twillpower, 2023-08-23:
Sorry, it has been resolved

closed 2023-08-25 by skarg

coskunyildirimcosel, 2023-10-16:
> Yığın patlamasından 20 önceliğiyle 10000 veya daha büyük bir değere ayarladım ve yığın taşması hala oluşuyor. void bacnet_task(void) { uint16_t pdu_len; BACNET_ADDRESS kaynağı; /* kaynak adresi */
> 
> ```
> /* handle the communication timer */
> if (mstimer_expired(&DCC_Timer))
> {
>     mstimer_reset(&DCC_Timer);
>     dcc_timer_seconds(DCC_CYCLE_SECONDS);
> }
> handler_cov_timer_seconds(1);
> // test
> HBD_LAMP_Is_People_Set(0, wy_is_people);
> count++;
> if(count >= 500)
> {
>     wy_is_people = !wy_is_people;
> }
> ```
> 
> // test /* mesajlaşmayı yönet */ pdu_len = datalink_receive(&src, &PDUBuffer[0], MAX_MPDU, 0); if (pdu_len) { // esp_log_buffer_hex("datalink_receive", &PDUBuffer[0], pdu_len); npdu_handler(&src, &PDUBuffer[0], pdu_len); } handler_cov_task(); }

Hello, may I know how you solved the stack overflow?