bacnet-stack/bacnet-stack #477: Does bacnet have a cov subscription recovery strategy?

Twillpower, 2023-08-25:
If device A initiates a permanent subscription to device B, but device B fails and restarts, how does device A know that it needs to re-initiate the subscription? Does device A need to delete the previous subscription, and if so, how to delete it?

skarg, 2023-08-25:
COV subscriptions with indefinite lifetimes (aka permanent) are discouraged.  See Q/A 7.6 in the [BTL Implementation Guildelines](https://bacnetglobal.org/wp-content/uploads/2022/08/BTL_Implementation_Guideline.pdf)

> 7.6 Do not subscribe for COV notifications with 'Lifetime' set to zero (indefinite lifetime)
Devices should not issue SubscribeCOV or SubscribeCOVProperty requests with the 'Lifetime' parameter
set to zero. (The value zero is prohibited for SubscribeCOVProperty requests.) The COV subscription list
is not guaranteed to remain through a device reset or power loss, or the device holding the subscription
could fail and be replaced. Worse, if the subscription list is permanently held and the subscribing device is
removed, the storage for the removed device's subscription remains allocated forever.

>SubscribeCOV requests should be issued with a non-zero 'Lifetime' parameter for some period (such as the
amount of time it is acceptable for the subscribing device to operate with out-of-date data), and the
subscription should be refreshed periodically.

Subscriptions include a subscriberProcessIdentifier that is unique per subscribing device. Any subsequent subscriptions to the same monitored object will replace the existing subscriptions with matching process-id and device source address. Example code from the library:
```
        if (COV_Subscriptions[index].flag.valid) {
            dest = cov_address_get(COV_Subscriptions[index].dest_index);
            if (dest) {
                address_match = bacnet_address_same(src, dest);
            } else {
                /* skip address matching - we don't have an address */
                address_match = true;
            }
            if ((COV_Subscriptions[index].monitoredObjectIdentifier.type ==
                    cov_data->monitoredObjectIdentifier.type) &&
                (COV_Subscriptions[index].monitoredObjectIdentifier.instance ==
                    cov_data->monitoredObjectIdentifier.instance) &&
                (COV_Subscriptions[index].subscriberProcessIdentifier ==
                    cov_data->subscriberProcessIdentifier) &&
                address_match) {
                existing_entry = true;
```

Twillpower, 2023-08-26:
thank you, sir

Twillpower, 2023-08-28:
```
typedef struct BACnet_Subscribe_COV_Data {
    uint32_t subscriberProcessIdentifier;
    BACNET_OBJECT_ID monitoredObjectIdentifier;
    bool cancellationRequest;   /* true if this is a cancellation request */
    bool issueConfirmedNotifications;   /* optional */
    uint32_t lifetime;  /* seconds, optional */
    bool covSubscribeToProperty;  /* true to use per-property subscription */
    BACNET_PROPERTY_REFERENCE monitoredProperty;
    bool covIncrementPresent;   /* true if present */
    float covIncrement; /* optional */
    BACNET_ERROR_CLASS error_class;
    BACNET_ERROR_CODE error_code;
    struct BACnet_Subscribe_COV_Data *next;
} BACNET_SUBSCRIBE_COV_DATA;
```
Is this lifetime in seconds? I set this value to 300, 600, and 6000, and it is difficult to receive a notify. The value I tested is about 3s to send a notify.
After I set the value of lifetime to 149130 * 60, I can receive notify within a period of time.
The following are the two notify logs received next to each other, and the actual time interval between them is not 905048 - 731141 seconds.
```
CCOV: Received Notification!
CCOV: PID=331 instance=331 WY_OBJECT_HBD_LAMP 0 time remaining=905048 seconds 
CCOV: proprietary 513  current value 0 
CCOV: Sending Simple Ack!
CCOV: Received Notification!
CCOV: PID=331 instance=331 WY_OBJECT_HBD_LAMP 0 time remaining=731141 seconds 
CCOV: proprietary 513  current value 1 
CCOV: Sending Simple Ack!

```

skarg, 2023-09-09:
The handler_cov_timer_seconds() is expecting to have a number of elapsed seconds called during a main loop or task or thread:
```
/** Handler to check the list of subscribed objects for any that have changed
 *  and so need to have notifications sent.
 * @ingroup DSCOV
 * This handler will be invoked by the main program every second or so.
 * This example only handles Binary Inputs, but can be easily extended to
 * support other types.
 * For each subscribed object,
 *  - See if the subscription has timed out
 *    - Remove it if it has timed out.
 *  - See if the subscribed object instance has changed
 *    (eg, check with Binary_Input_Change_Of_Value() )
 *  - If changed,
 *    - Clear the COV (eg, Binary_Input_Change_Of_Value_Clear() )
 *    - Send the notice with cov_send_request()
 *      - Will be confirmed or unconfirmed, as per the subscription.
 *
 * @note worst case tasking: MS/TP with the ability to send only
 *        one notification per task cycle.
 *
 * @param elapsed_seconds [in] How many seconds have elapsed since last called.
 */
void handler_cov_timer_seconds(uint32_t elapsed_seconds)
```

Twillpower, 2023-09-13:
Thank you, I understand, but the parameter of this function is uint32_t, and the minimum value is 1. If the status changes, but the notification is sent after 1s, the delay will definitely be greater than 1s. This will affect real-time performance.

skarg, 2023-09-13:
The value of timeout is for the subscription timeout, not for the COV itself.  COV will occur at real-time speed during the timeout period, and will stop when the timeout expires.

skarg, 2023-09-13:
handler_cov_timer_seconds() is only handling the timeout of the subscription, which doesn't affect COV Notification speed.

handler_cov_fsm() or handler_cov_task() is the function that handles the COV notifications and can be called as fast as the CPU allows. You can rewrite the h_cov.c if you require to have better event driven performance for your system - it is just an example of basic COV functionality.

Twillpower, 2023-09-18:
```
void my_task()
{
    while(1)
    {
        handler_cov_timer_seconds(1);
        wy_is_people = wy_get_is_people();
        HBD_LAMP_Is_People_Set(0, wy_is_people);
        vTaskDelay(10 / portTICK_PERIOD_MS);
        // or  vTaskDelay(1000 / portTICK_PERIOD_MS);  ?
    }
}
```
emmm, what I mean is probably similar to this. If you let it delay 1s, wy_is_people will be monitored once every second. If you let it delay 10ms, the monitoring will be more timely, but handler_cov_timer_seconds is not 1s from the last call. (Am I understanding it wrong, or is there a better solution)

Twillpower, 2023-09-19:
Thanks, I roughly got it, I'm building cov into a separate task now.
```
void wy_updata_cov(void)
{
    AIR_INFO m_air_info_data = {0};
    struct mstimer COV_Timer;
    mstimer_set(&COV_Timer, 1 * 1000);
    while(1)
    {
    if (mstimer_expired(&COV_Timer)) {
        mstimer_reset(&COV_Timer);
        handler_cov_timer_seconds(mstimer_interval(&COV_Timer)/1000);
    }
        m_air_info_data = wy_get_air_info();
        AIR_Info_Set(0, m_air_info_data);
        handler_cov_task();
        vTaskDelay(10 / portTICK_PERIOD_MS);
    }
}
```

BACnetEd, 2023-09-20:
Hi

I would highly recommend that, while you do so, you make the whole COV detection system event-driven.