bacnet-stack/bacnet-stack #478: make windows shared library errors

badboy-huaqiao, 2023-08-25:
Hi all, I got the below errors when I wanted to make shared library of windows.

`cd apps\lib` and` make lib`, got the errors:

`cc -c -Wall -Wmissing-prototypes  -Os  -IE:/my_workspace/bacnet/bacnet-stact-1.1.1/src -ffunction-sections -fdata-sections E:/my_workspace/bacnet/bacnet-stact-1.1.1/src/bacnet/bacpropstates.c -o E:/my_workspace/bacnet/bacnet-stact-1.1.1/src/bacnet/bacpropstates.o
process_begin: CreateProcess(NULL, cc -c -Wall -Wmissing-prototypes -Os -IE:/my_workspace/bacnet/bacnet-stact-1.1.1/src -ffunction-sections -fdata-sections E:/my_workspace/bacnet/bacnet-stact-1.1.1/src/bacnet/bacpropstates.c -o E:/my_workspace/bacnet/bacnet-stact-1.1.1/src/bacnet/bacpropstates.o, ...) failed.
make (e=2):
Makefile:121: recipe for target 'E:/my_workspace/bacnet/bacnet-stact-1.1.1/src/bacnet/bacpropstates.o' failed
make: *** [E:/my_workspace/bacnet/bacnet-stact-1.1.1/src/bacnet/bacpropstates.o] Error 2`

skarg, 2023-08-25:
apps/lib/Makefile usually depends on settings in apps/Makefile.  You can invoke apps/lib from root Makefile:
`make lib`
or from apps makefile
`make -C apps lib`

badboy-huaqiao, 2023-08-25:
@skarg thanks for your replay, but I still got the below error, I try to use other branch like 0.8.6 or 1.1.2, still can not work:

`E:\my_workspace\bacnet-stack> make lib
C:/TDM-GCC-64/bin/make -s -C apps lib
process_begin: CreateProcess(NULL, uname -s, ...) failed.
process_begin: CreateProcess(NULL, cc -c -Wall -Wextra -Wfloat-equal -Wconversion -Wparentheses -pedantic -Wunused-parameter -Wunused-variable -Wreturn-type -Wunused-function -Wreturn-type -Wunused-value -Wswitch-default -Wuninitialized -Winit-self -Wno-sign-conversion -Wno-conversion -Wno-sign-compare -Wno-long-long -Wno-implicit-fallthrough -Wno-attributes -Os -DBACDL_BIP=1 -DBBMD_ENABLED=1 -DBBMD_CLIENT_ENABLED -DWEAK_FUNC= -D_NO_OLDNAMES -DPRINT_ENABLED=1 -DBACAPP_ALL -DBACFILE -DINTRINSIC_REPORTING -DBACNET_TIME_MASTER -DBACNET_PROPERTY_LISTS=1 -DBACNET_PROTOCOL_REVISION=24 -IE:/my_workspace/bacnet-stack/src -IE:/my_workspace/bacnet-stack/ports/win32 -std=gnu89 -ffunction-sections -fdata-sections -Wall -Wextra -Wfloat-equal -Wconversion -Wparentheses -pedantic -Wunused-parameter -Wunused-variable -Wreturn-type -Wunused-function -Wreturn-type -Wunused-value -Wswitch-default -Wuninitialized -Winit-self -Wno-sign-conversion -Wno-conversion -Wno-sign-compare -Wno-long-long -Wno-implicit-fallthrough -Wno-attributes -Os -DBACDL_BIP=1 -DBBMD_ENABLED=1 -DBBMD_CLIENT_ENABLED -DWEAK_FUNC= -D_NO_OLDNAMES -DPRINT_ENABLED=1 -DBACAPP_ALL -DBACFILE -DINTRINSIC_REPORTING -DBACNET_TIME_MASTER -DBACNET_PROPERTY_LISTS=1 -DBACNET_PROTOCOL_REVISION=24 -IE:/my_workspace/bacnet-stack/src -ffunction-sections -fdata-sections -DBACDL_BIP=1 E:/my_workspace/bacnet-stack/src/bacnet/bacpropstates.c -o E:/my_workspace/bacnet-stack/src/bacnet/bacpropstates.o, ...) failed.
make (e=2):
Makefile:121: recipe for target 'E:/my_workspace/bacnet-stack/src/bacnet/bacpropstates.o' failed
make[2]: *** [E:/my_workspace/bacnet-stack/src/bacnet/bacpropstates.o] Error 2
Makefile:211: recipe for target 'E:/my_workspace/bacnet-stack/apps/lib/libbacnet.a' failed
make[1]: *** [E:/my_workspace/bacnet-stack/apps/lib/libbacnet.a] Error 2
Makefile:46: recipe for target 'lib' failed
make: *** [lib] Error 2`

badboy-huaqiao, 2023-08-25:
@skarg Hi, this bacnet stack you made is really good, but unfortunately, I'm not a C developer even though I know a little C programming, I‘m good at Golang, so I want to call the C library with cgo module of Golang , so do you think that’s possible? 
or do you have a plan to make a golang client wrapper?

skarg, 2023-08-25:
I run my Windows library Makefile builds under MINGW32 or MINGW64 (or git bash with a MSYS/MINGW and some .bashrc exports).  If you are trying to build outside of those shells, perhaps using CMake would be a better choice.  CMake is a different build system than Makefiles, and can [generate a large variety of build systems](https://cmake.org/cmake/help/latest/manual/cmake-generators.7.html#manual:cmake-generators(7)) and this is easily done in Visual Studio Code with CMake extension.

The library is also tested with CMake under Windows/Linux/MacOS - see the [github actions](https://github.com/bacnet-stack/bacnet-stack/blob/master/.github/workflows/main.yml).

In the error message above, I see the NULL near `uname - s`, which I just recently added to the apps/Makefile to help with OS detection under MINGW, which is likely causing that error. Thanks for reporting that - I'll get it fixed.

Have you looked at a BACnet-Go specific project? Craig contacted me about 5 years ago about it:
[https://github.com/stitchcula/bacnet-go](https://github.com/stitchcula/bacnet-go)

badboy-huaqiao, 2023-08-25:
thanks for your help, I use the git bash, but still not work yet, by the way, I don't know how and what to do with `some .bashrc exports` you mentioned above,  here is the errors:
```
MINGW64 /e/golang_workspace/bacnet-stack (bacnet-stack-1.1.2)

$ make lib
D:/mingw64/bin/make -s -C apps lib
process_begin: CreateProcess(NULL, cc -c -Wall -Wextra -Wfloat-equal -Wconversion -Wparentheses -pedantic -Wunused-parameter -Wunused-variable -Wreturn-type -Wunused-function -Wreturn-type -Wunused-value -Wswitch-default -Wuninitialized -Winit-self -Wno-sign-conversion -Wno-conversion -Wno-sign-compare -Wno-long-long -Wno-implicit-fallthrough -Wno-attributes -Os -DBACDL_BIP=1 -DBBMD_ENABLED=1 -DBBMD_CLIENT_ENABLED -DWEAK_FUNC= -D_NO_OLDNAMES -DPRINT_ENABLED=1 -DBACAPP_ALL -DBACFILE -DINTRINSIC_REPORTING -DBACNET_TIME_MASTER -DBACNET_PROPERTY_LISTS=1 -DBACNET_PROTOCOL_REVISION=24 -IE:/golang_workspace/bacnet-stack/src -IE:/golang_workspace/bacnet-stack/ports/win32 -std=gnu89 -ffunction-sections -fdata-sections -Wall -Wextra -Wfloat-equal -Wconversion -Wparentheses -pedantic -Wunused-parameter -Wunused-variable -Wreturn-type -Wunused-function -Wreturn-type -Wunused-value -Wswitch-default -Wuninitialized -Winit-self -Wno-sign-conversion -Wno-conversion -Wno-sign-compare -Wno-long-long -Wno-implicit-fallthrough -Wno-attributes -Os -DBACDL_BIP=1 -DBBMD_ENABLED=1 -DBBMD_CLIENT_ENABLED -DWEAK_FUNC= -D_NO_OLDNAMES -DPRINT_ENABLED=1 -DBACAPP_ALL -DBACFILE -DINTRINSIC_REPORTING -DBACNET_TIME_MASTER -DBACNET_PROPERTY_LISTS=1 -DBACNET_PROTOCOL_REVISION=24 -IE:/golang_workspace/bacnet-stack/src -ffunction-sections -fdata-sections -DBACDL_BIP=1 E:/golang_workspace/bacnet-stack/src/bacnet/bacpropstates.c -o E:/golang_workspace/bacnet-stack/src/bacnet/bacpropstates.o, ...) failed.
make (e=2): ϵͳ▒Ҳ▒▒▒ָ▒▒▒▒▒ļ▒▒▒
make[2]: *** [Makefile:121: E:/golang_workspace/bacnet-stack/src/bacnet/bacpropstates.o] Error 2
make[1]: *** [Makefile:235: E:/golang_workspace/bacnet-stack/apps/lib/libbacnet.a] Error 2
make: *** [Makefile:46: lib] Error 2
``` 

> Have you looked at a BACnet-Go specific project? Craig contacted me about 5 years ago about it:
> https://github.com/stitchcula/bacnet-go

yes, I have found it before, but unfortunately, the author didn't update it anymore, it seems like nothing have been implemented, and I think that's not an easy work implementing the bacnet protocol with Golang from scratch.
so I think the way using cgo module of Golang call your C stack library is more better and easier, but I'm not sure, there is two ways Golang call C, use the C source file or C shared library, I try to use C source file but there are too many header files and path of them to import, so I want to try the C shared library. 

Looking forward your updates.
thanks again, best regards!

skarg, 2023-08-25:
My .bashrc in git bash looks like this:
```
$ cat ~/.bashrc
alias make=mingw32-make.exe
PATH=/c/msys64/mingw32/bin:$PATH
export MAKE=mingw32-make.exe
export CC=gcc.exe
export OBJCOPY=objcopy.exe
export AR=ar.exe
export NM=nm.exe
export STRIP=strip.exe
export SIZE=size.exe
```
The apps/Makefile has some hints about building with MinGW:
```
ifeq (${BACNET_PORT},win32)
# winget install --id=MSYS2.MSYS2  -e
# pacman -Syu mingw-w64-i686-toolchain
# add alias make=mingw32-make.exe to .bashrc
...
```

closed 2023-09-09 by skarg