bacnet-stack/bacnet-stack #486: read remote prop and return value same function

cyberumut, 2023-09-06:
https://github.com/bacnet-stack/bacnet-stack/blob/bacnet-stack-0.8/demo/readprop/main.c

im doing read remote analog input value but returned value this MyReadPropertyAckHandler. 

how can i same function read and can i get returned value ?

for example :

int readprop(int target address,int prop){

int y = Send_Read_Property_Request(Target_Device_Object_Instance,
                    Target_Object_Type, Target_Object_Instance,
                    Target_Object_Property, Target_Object_Index);

int returnedvalue = y;

return returnedvalue;

}

skarg, 2023-09-06:
There is a BACnet ReadProperty and WriteProperty client example and some core code to support the state machine in [bac-rw.c](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/client/bac-rw.c) file. The [server-client](https://github.com/bacnet-stack/bacnet-stack/blob/master/apps/server-client/main.c) example shows how that library is integrated into a working application.

cyberumut, 2023-09-06:
> There is a BACnet ReadProperty and WriteProperty client example and some core code to support the state machine in [...]

my bacnet server is already running in the background..

I just want to return the value I want to read in some functions, how can I do that?
While communicating with the server and other devices in the background, I want to read the value from the remote device with the readpropremote() function.

skarg, 2023-09-06:
Add [bac-rw.c](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/client/bac-rw.c) and [bac-data.c](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/client/bac-data.c) to your server build. 

Add to your server main-loop a call to bacnet_data_task();

Add to your server init a call to bacnet_data_init();

See [bac-task.c](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/client/bac-task.c) as an example of where to place the function calls.

bac-data.c is an example of ReadProperty for Analog, Binary, and Multi-State object present-value properties, Use the functions to read the properties:
```
bool bacnet_data_analog_present_value(uint32_t device_id,
    BACNET_OBJECT_TYPE object_type,
    uint32_t object_instance,
    float *float_value);

bool bacnet_data_multistate_present_value(uint32_t device_id,
    uint16_t object_type,
    uint32_t object_instance,
    uint32_t *unsigned_value);

bool bacnet_data_binary_present_value(uint32_t device_id,
    uint16_t object_type,
    uint32_t object_instance,
    bool *bool_value);
```
Usually the first time function call will result in return of false as the request gets queued and sent and will be waiting for a response or timeout.  When the call is true, then the data has arrived.  The task is configured to poll every 60s, but poll time can be configured with bacnet_data_poll_seconds_set().

cyberumut, 2023-09-06:
> Add [bac-rw.c](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/client/bac-rw.c) and [...]

is this not working bacnet 0.8.5 ? 

i added my project  but always returned value 0.00000

my server code : `void server(int id)
{

    Init_Service_Handlers();
    Device_Set_Object_Instance_Number(id);

    BACNET_ADDRESS src = {0}; /* address where message came from */
    uint16_t pdu_len = 0;
    unsigned timeout = 100; /* milliseconds */
    unsigned max_apdu = 0;
    time_t elapsed_seconds = 0;
    time_t last_seconds = 0;
    time_t current_seconds = 0;
    time_t timeout_seconds = 0;
    bool found = false;
    /* who is  start***/
    time_t total_seconds = 0;

    datalink_get_broadcast_address(&dest);

    atexit(datalink_cleanup);
	bacnet_data_init();
	bacnet_data_poll_seconds_set(10);
    last_seconds = time(NULL);
    timeout_seconds = (apdu_timeout() / 1000) * apdu_retries();

  
    Send_I_Am(&Handler_Transmit_Buffer[0]);

    
    Send_WhoIs_To_Network(&dest, Target_Object_Instance_Min,
                          Target_Object_Instance_Max);
   
    while (!serverstart) 
    {
       
        current_seconds = time(NULL);

        
        if (current_seconds != last_seconds)
            tsm_timer_milliseconds((uint16_t)((current_seconds -
                                               last_seconds) *
                                              1000));

       
        pdu_len = datalink_receive(&src, &Rx_Buf[0], MAX_MPDU, timeout);

       
        if (pdu_len)
        {
            npdu_handler(&src, &Rx_Buf[0], pdu_len);
        }

      
        last_seconds = current_seconds;

      
		bacnet_data_task();
   }
	

}`

cyberumut, 2023-09-06:
float amkfloat= 0.0;

	 BACNET_OBJECT_TYPE tiptip= OBJECT_ANALOG_OUTPUT;
	
	bacnet_data_binary_present_value(31,
    tiptip,
    2,
    &amkfloat);
	printf("@@@@@@@@@@@@@@@@@@@ --- %f \n" ,amkfloat);

cyberumut, 2023-09-07:
> Add [bac-rw.c](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/client/bac-rw.c) and [...]

is this working for 0.8.5 version ?

skarg, 2023-09-07:
for: OBJECT_ANALOG_OUTPUT, OBJECT_ANALOG_INPUT, OBJECT_ANALOG_VALUE
float data type
use bacnet_data_analog_present_value() function

for: OBJECT_BINARY_OUTPUT, OBJECT_BINARY_INPUT, OBJECT_BINARY_VALUE
bool data type
use bacnet_data_binary_present_value() function

for: OBJECT_MULTI_STATE_OUTPUT, OBJECT_MULTI_STATE_INPUT, OBJECT_MULTI_STATE_VALUE
uint32_t data type
use bacnet_data_multistate_present_value() function

cyberumut, 2023-09-07:
I want to read analog output 2

I am using the bacnet_data_analog_present_value function

but always returned 0.00000

skarg, 2023-09-07:
BACnet server applications will likely need to define at least 1 slot each for:
```
MAX_ADDRESS_CACHE
MAX_TSM_TRANSACTIONS
```
Are those defined in your project settings?

Does a Wireshark capture indicate the ReadProperty are being sent?

The [server-client example app](https://github.com/bacnet-stack/bacnet-stack/tree/master/apps/server-client) is using the same files.  Perhaps it can serve as a guide?  Is the example application working?

cyberumut, 2023-09-07:
yeah bacpoll.exe working. but iam using 0.8.5 version with esp32 
When I use the example server client code, it doesn't go inside the if
```
 for (;;) {
        bacnet_task();
        if (mstimer_expired(&print_value_timer)) {.....
```

skarg, 2023-09-07:
mstimer module needs the port of mstimer_now() to work - is mstimer_now() implemented and counting? You could use a seconds counter instead of mstimer module. The calls to bacnet_data_analog_present_value() function don't really need a timer, but for the example application I made the timer the same as the poll frequency.

I don't think there is something missing in 0.8.5 that would prevent bac-rw.c module from working, but debugging might help narrow scope.

cyberumut, 2023-09-07:
> mstimer module needs the port of mstimer_now() to work - is mstimer_now() implemented and counting? You could use a seconds counter instead of mstimer module. The calls to [...]

thanks steve

closed 2023-09-07 by cyberumut